sstsimulator/sst-elements #1485: scatter not working for any other root except 0

khodarkhashi, 2020-05-18:
I am trying to use enQ_scatter for my simulation along a custom communicator. But it gets into segfault. When I use GroupWorld and 0 as root it works fine. My guess is scatter only works for GroupWorld and root as 0. Since using a custom communicator does not ensure the root as 0 , that results into a segfault.

jjwilke, 2020-05-18:
Can you post a minimal reproducer code and Python file?

Custom communicators are probably more heavily tested in SST/macro skeletons - so there are other options to get you started if we can't get this working.

nmhamster, 2020-07-12:
@mjleven - this sounds like a FireFly bug?

nmhamster, 2020-07-16:
Please see PR: #1511 - thank you to Mike Levenhagen for working through this issue.

nmhamster, 2020-10-13:
Issued closed with #1511

closed 2020-10-13 by nmhamster