sstsimulator/sst-elements #1490: FATAL: l1_cache_1:memlink (MemNICBase) cannot find a destination for address

afranques, 2020-06-02:
Hello, this is an issue (somewhat) related to #1489, happening in SST v10.0.0.

I am trying to configure this (simple?) mesh architecture:

[image]

The script runs fine when there's only 1 tile (`n_tiles=1`). However, when I run it with either 2 (`n_tiles=2`) or 4 tiles (`n_tiles=4`), I am getting error: `FATAL: l1_cache_1:memlink (MemNICBase) cannot find a destination for address 1261442264237737216`. The output around that line looks like:
```
I am working on: rtr.0.0 phase: 5
I am working on: rtr.0.0 phase: 5
core_0: 10 Issued Read for address 0xb16b8b0e35980500
core_1: 10 Issued Read for address 0xb16b8b0e35980500
E: 1000                 2                    l1_cache_0           Event:New     (<13,0> GetS Src: core_0:memory Dst: l1_cache_0 Rq: l1_cache_0 Flags: [] Addr: 0xb16b8b0e35980500 (G) Data: F VA: 0x0 IP: 0x0 Size: 4 Prf: F)
M: 1000                                      l1_cache_0           MSHR:InsEv    0xb16b8b0e35980500 Sz: 1      (<13,0>, pos=0)
C: 1000                 2                    l1_cache_0           Alloc         0xb16b8b0e35980500
C: 1000                 2                    l1_cache_0           GetS          0xb16b8b0e35980500 <13,0>          I      IS     Forward    (miss)          LLSC: N Lock: (0,0)
E: 1000                 2                    l1_cache_1           Event:New     (<14,0> GetS Src: core_1:memory Dst: l1_cache_1 Rq: l1_cache_1 Flags: [] Addr: 0xb16b8b0e35980500 (G) Data: F VA: 0x0 IP: 0x0 Size: 4 Prf: F)
M: 1000                                      l1_cache_1           MSHR:InsEv    0xb16b8b0e35980500 Sz: 1      (<14,0>, pos=0)
C: 1000                 2                    l1_cache_1           Alloc         0xb16b8b0e35980500
FATAL: l1_cache_1:memlink (MemNICBase) cannot find a destination for address 12784464860366243072
Known destinations: 
FATAL: l1_cache_1:memlink (MemNICBase) cannot find a destination for address 12784464860366243072
Known destinations: 
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3ea) [0x4f09aa]
    1 : /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy10MemNICBase21findTargetDestinationB5cxx11Em+0x696) [0x7f286171aed6]
    2 : /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy19CoherenceController14forwardMessageEPNS0_8MemEventEjmPSt6vectorIhSaIhEE+0x1c1) [0x7f28617517d1]
    3 : /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy6MESIL110handleGetSEPNS0_8MemEventEb+0x224) [0x7f28617662d4]
    4 : /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy5Cache12processEventEPNS0_12MemEventBaseEb+0x47f) [0x7f286167b50f]
    5 : /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmemHierarchy.so(_ZN3SST12MemHierarchy5Cache9clockTickEm+0x5d1) [0x7f286167bd61]
    6 : sst(_ZN3SST5Clock7executeEv+0xe6) [0x4bacb6]
    7 : sst(_ZN3SST10Simulation3runEv+0x1e7) [0x4fb707]
    8 : sst() [0x4a9c12]
    9 : sst(main+0xeca) [0x49a1ea]
   10 : /lib64/libc.so.6(__libc_start_main+0xf5) [0x7f2877a56555]
   11 : sst() [0x4a94ec]
```

Additionally, for any number of tiles bigger than 4, like for `ntiles=8`, or `ntiles=16`, instead of that I get error: `Signal code: Address not mapped (1)`. The output around that line looks like:
```
I am working on: rtr.3.1 phase: 1
I am working on: rtr.3.1 phase: 1
I am working on: rtr.3.1 phase: 1
I: l1_cache_7             Event:Init      (<6,0> NULLCMD Src: core_7:memory Dst: None Rq: None Flags: [] InitCmd: Coherence Type: 0 Inclusive: false LineSize: 0 Tracks presence: false)
l1_cache_7 processInitCoherenceEvent. Result is: LL ? N, silentEvict ? Y, WBClean ? Y, sendWBAck ? N, recvWBAck ? N
I am working on: rtr.2.1 phase: 1
I am working on: rtr.2.1 phase: 1
*** Process received signal ***
Signal: Segmentation fault (11)
Signal code: Address not mapped (1)
Failing at address: 0x50
[ 0] /lib64/libpthread.so.0(+0xf630)[0x7f44a1389630]
[ 1] sst(_ZNK3SST5UnitseqERKS0_+0x4)[0x50df74]
[ 2] sst(_ZNK3SST11UnitAlgebragtERKS0_+0x27)[0x510157]
[ 3] /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin11PortControl4initEj+0xdf)[0x7f448983c31f]
[ 4] /home/franque2/workspace/sst-widir/local/sstelements-10.0.0/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin9hr_router4initEj+0x6d8)[0x7f448981d548]
[ 5] sst(_ZN3SST10Simulation10initializeEv+0x74)[0x4f9824]
[ 6] sst[0x4a9bf2]
[ 7] sst(main+0xeca)[0x49a1ea]
[ 8] /lib64/libc.so.6(__libc_start_main+0xf5)[0x7f449f82f555]
[ 9] sst[0x4a94ec]
 *** End of error message ***
Segmentation fault (core dumped)
```

My configuration file looks like this (remove the `.txt` extension if you download it): [baseline_mesh_trivialcpu_github.py](https://github.com/sstsimulator/sst-elements/files/4713953/baseline_mesh_trivialcpu_github.py.txt)

This configuration basically consists of a certain number of tiles (defined by the `n_tiles` variable in `line 18`), where each tile consists of:
 - One core (of type `memHierarchy.trivialCPU`)
 - One private L1
 - One slice of the shared L2 (shared between all tiles)
 - One memory controller of type `memHierarchy.simpleDRAM` **(only in the first tile)** 
 - One router (of type `merlin.hr_router`)
 
Additionally:
 - The L2 is inclusive of the L1, and contains directory information
 - We use MESI coherence
 - We use a Melin mesh NoC, with one router per tile

Things that I have tried to fix it:
 - I tried replacing Merlin mesh for Kingsley, but I got the same error. 
 - I tried creating two links (instead of one) between each pair of routers, but nothing changed
 - I tried adapting the Merlin mesh example (`ex9.py`) in the [sst-tutorials repo](https://github.com/sstsimulator/sst-tutorials/tree/master/exercises/ex9), but even that one doesn't seem to work anymore, as it keeps throwing `FATAL: Merlin: hr_router requires topology to be specified in input file`, even when the file clearly contains `topology: merlin.mesh` in it
 - I tried adapting the Merlin torus example (`ariel_snb_mlm.py`) in the [Ariel tests folder](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/ariel/frontend/simple/examples/stream/ariel_snb_mlm.py), but as soon as I changed the torus shape from 1D to 2D, it threw error `FATAL:  SST Core: Error:  TimeConverter creation requires a time unit (s or Hz), 0 b/s was passed to call`

Things that maybe are producing the error:
 - It seems like the L1 can't reach the home slice of the address it's trying to access. So, could it be I'm not connecting the L1 and/or the L2 slices properly to the routers?
 - Could it be that I'm not setting the `group` number of each component properly? Maybe, but I did it in a similar way as in [`testDistributedCaches.py`](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/tests/testDistributedCaches.py)
 - Could it be that because I only have 1 memory controller and it's in only 1 of the tiles, the other tiles can't find it? This shouldn't be an issue, since the request should be able to traverse the mesh. Although maybe I didn't configure properly the link between the router and the mem controller?
 - Could it be that I'm not connecting the routers properly? Maybe, but I did it similarly as in `testKingsley.py`, so I doubt it's this
 - Could it be that because I'm leaving an empty local port in all routers except the one in `tile0` (for the memory controller), I'm messing up with the routing algoritm?

I would really appreciate if anyone took the time to check my architecture diagram and/or configuration file, and point out any potential design flaw. Thanks in advance.

gvoskuilen, 2020-06-02:
Thanks for posting this, it's very thorough which is helpful! I am looking at your setup and hopefully will have an answer soon.

gvoskuilen, 2020-06-03:
@afranques Try making the following fixes in memHierarchy:
1. In the core params, add {"memSize" : 173741824}. (For 1GB, adjust as needed to match the simulated memory size)
2. This is a typo and isn't causing an error since it's default, but change "coherence_policy" to "coherence_protocol" in the cache parameters
3. In "mem_ctrl_params", add {"addr_range_start" : 0, "addr_range_end" : 1073741824-1}. This declares which addresses are valid at the memory.

Generally those steps would fix this error, but the error message also shows that the caches haven't found any destinations at all which points to an issue in the network configuration. I would recommend using the Merlin Python module to generate a mesh for you. To do that, configure like this (replace XX with the right values). The module generally prompts you if there is a missing parameter it needs.

```
import sst
from sst.merlin import *

topo = topoMesh()
sst.merlin._params["link_lat"] = XX

# Set parameters the same way for "link_bw", "xbar_bw", "flit_size", "input_latency", 
# "output_latency", "input_buf_size", "output_buf_size", "num_dims", 
# "mesh:width" (this is number of links in each dimension separated with 'x' such as "1x1"), 
# "mesh_shape" (such as "3x3", "2x4", etc.), "mesh:local_ports" (3 for your config)

# Merlin will ingest the params and prompt if it didn't get something 
topo.prepParams() 

# Create an endpoint function to generate the components that need to be connected 
# to each local port (e.g., MemBuilder, L1Builder, L2Builder)
class MemBuilder:
     def build(self, portnum, extraKeys):
        # Build memory controller/backend/etc
        return (memlink, "port", "latency")

# Create an empty function for unpopulated ports
class NullBuilder:
    def build(self, portnum, extraKeys):
        pass

# Create a dispatch function that takes the port num and returns the right builder 
# portnum will be 0 to nodes*localports-1
def setNode(portnum):
    # Figure out which localport/node you're building and return the right builder

topo.setEndPointFunc(setNode)
topo.build()
```

afranques, 2020-06-03:
Thank you very much for your reply, @gvoskuilen. I'm debugging another component now, but as soon as I can I will switch back to the network config file and try this.

gvoskuilen, 2020-08-17:
@afranques Did this solve the issue? If so, can you close it? Thank you.

afranques, 2020-09-07:
Yes it did. I closed the issue. Thank you very much for the help!

closed 2020-09-07 by afranques