sstsimulator/sst-elements #1497: simulator time base access

gunney1, 2020-06-13:
How to access the simulator time base from components
--------------------------

I set the simulator time base in the python script using sst.setProgramOption("timebase", "1 ps").  How I know in the C++ code if it was "1 ps" or "1 ns", or whatever?  I couldn't find anything in the Simulator or TimeLord classes that may tell me.

The reason I want to do this is that some parameters passed in are in specific units, for example, seconds.  I need to convert them into the units set by the Python script?  I guess I can send this info in through the component's Params argument.  I just thought that maybe this information is being kept somewhere behind a public interface.  Thanks.

jjwilke, 2020-06-14:
Could you give an example of the conversion you need to do? And you need this in C++ or Python? The preferred way is usually to use the SST::UnitAlgebra class.

feldergast, 2020-06-15:
A couple points:

In general, you do not need to set the timebase.  It is set to 1ps by default and will work for the vast majority of simulations.  The only time you may need to change it is if you are working with very high frequency events 10-100's of GHz, in which case you would need to set it to a smaller number, or are working with very low frequency events (on the order of seconds) and need to run for longer than ~250 simulated days.

It is possible to get the time base by using TimeLord::getTimeBase.  This will return a UnitAlgebra object with the core timebase.

Depending on what you want to do, you may not need to do conversions.  If you're parameters are coming in with units, you can simply pass them to the various time functions (registerClock, registerTimeBase, getTimeConverter, etc.); all of these functions can take a string which include units or a UnitAlgebra object.  If you need to convert for your own use, then using the UnitAlgebra object is the way to go.  It is currently only available in the c++ classes, but will soon be added to the python.

gunney1, 2020-06-15:
Thanks for the instructions to use UnitAlgebra.  That seems to be what I need.

closed 2020-06-15 by gunney1