sstsimulator/sst-elements #1506: Dragonfly intergroup link and number of group can not be set as 1

khodarkhashi, 2020-06-23:
I have been using dragonfly topologies and I believe the order in the shape parameters are like this - 

hosts per router : number of routers per group : inter group links : number of groups

If I try with intergroup links = 1 and number of groups = 1 , it gives a floating point exception. 

Suppose I want to use only 1 node with 8 cores in a dragonfly topology. Would you suggest me on how to set the params for these kind of configs? I am using the defaultSim.py file so configure the parameters and using it like this - 

def getWorkFlow( defaults ):
    workFlow = []
    motif = dict.copy( defaults )
    motif['cmd'] = "Init"
    workFlow.append( motif )

    motif = dict.copy( defaults )
   # motif['cmd'] = "MY MOTIF CALL"     
   
    workFlow.append( motif )

    motif = dict.copy( defaults )
    motif['cmd'] = "Fini"
    workFlow.append( motif )

        # numNodes = 0 implies use all nodes on network
    numNodes = 0
    **numCores = 8**

    return workFlow, numNodes, numCores

def getNetwork():

        #platform = 'chamaPSM'
        #platform = 'chamaOpenIB'
        #platform = 'bgq'
        platform = 'default'

        #topo = ''
        #shape = ''
        **topo = 'dragonfly'
        shape = '1:1:1:1'**

        return platform, topo, shape
Thanks in advance.

feldergast, 2020-06-24:
Dragonfly topology requires at least two groups.  If you want to simulate a single router, use the "Single" topology.  If you want a single group, this is just a 1D hyperX, so you would use a hyperX topology.

khodarkhashi, 2020-06-24:
Thank you. Could you help me with "numNodes" and "numCores" parameters? What do they actually represent?

mjleven, 2020-07-01:
In the case of the defaultSim.py numNodes represents the number of nodes the motifs will run on. This number could be less than the number of nodes for the network topology. numCores will be the number of Ember instances on a given node. An Ember instance will run a instance of the motifs. For example if the number of nodes is 8 and the numCores is 8 there would be 64 instances of Ember. Each instance of Ember is an MPI rank.

khodarkhashi, 2020-07-06:
Thank you for your kind response. As per your suggestion I tried to run the "FFT3D" motif using the following values - 

numNodes = 4
numCores = 1
topo = 'torus'
shape = '2x2x2'

So in the topology there are 8 nodes and we want to use 4 of them using numNodes variable. But unfortunately this creates an error(given below). When I use numNodes = 8 it works fine. Any value of numNodes < 8 creates this error (I believe this is coming from EmberEP.py). I used different combinations of numNodes and numCores like numNodes = 4 , numCores = 2 and numNodes = 2 , numCores = 4. All of them gives same error. 

I am using the 9.1.0 version of SST. Is this something which you fixed in 10.0.0? 

The error is this - 
  File "/global/cscratch1/sd/afibuzza/SST_9.1.0/sst-elements/src/sst/elements/ember/test/emberLoad.py", line 510, in <module>
    topo.build()
  File "", line 195, in build
    
  File "EmberEP.py", line 187, in build
    os.addParams( {'netMapId': self.nidMap[ nodeID ] } )
IndexError: string index out of range
FATAL: SSTPythonModel Execution of model construction function failed.
FATAL: SSTPythonModel Execution of model construction function failed.
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3d9) [0x4e0609]
    1 : sst(_ZN3SST4Core24SSTPythonModelDefinition17createConfigGraphEv+0x132) [0x53f0c2]
    2 : sst(main+0x394) [0x492094]
    3 : /lib64/libc.so.6(__libc_start_main+0xea) [0x2aaaac7db34a]
    4 : sst(_start+0x2a) [0x4a069a]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.

Thank you for your response again.

DanieleDeSensi, 2020-12-17:
Do you have any update on this? I'm also trying to run a test with numNodes smaller than the number of nodes in the topology and I get the same error.

feldergast, 2020-12-17:
In the case of numNodes < 8, you have nodes in the network that have not been allocated a job.  If you don't want to use a node, you need to use the "Null" motif to indicate that.  If a node does not have any jobs assigned to it, then it will fail when trying to initialize things (as seen in the above error).

mdafibuzzaman, 2020-12-17:
As Scott suggested, from the input defaultSim.py we can not use use numNodes < "node counts in topology". For workaround I modified my motif in such a way that it only uses certain nodes and not all the nodes. I was not sure where/how to assign individual nodes as NULL.

feldergast, 2020-12-17:
Using more than on "job" requires the use of a load file.  There is an example in sst/elements/ember/tests.

The files:

qos-dragonfly.sh
qos-fattree.sh
qos-hyperx.sh

Are scripts which use emberload.py with a load file.  The load file itself is qos.load.  One of the job descriptions is copied below with comments:

# JOB_ID must be unique for each job, otherwise you will get undefined behavior
[JOB_ID] 1.

# This gives the list of nodes to use.  You can just use a comma separated list, or on 
# of the nidlist generators that are provided.  You almost certainly don't want the QOS 
# version used here, but other ones are found in /sst/elements/ember/test (they are 
# the python files that start with generateNidList, you can look in the files to see 
# exactly what they do).
[NID_LIST] generateNidList=generateNidListQOS(512,0,random,128,1)

# You can pass parameters into the job here.  These are for qos and will likely not
# be needed for what you are doing.  Often, not extra params need to be passed
# here since they are typically passed in the MOTIF section
[PARAM] merlin:portcontrol:output_arb=merlin.arb.output.qos.multi
[PARAM] merlin:portcontrol:arbitration:qos_settings=[25,25,25,25]

# This just lists the order of motifs to run.  For an empty job, just use:
# [MOTIF] Null
[MOTIF] Init
[MOTIF] Sweep3D pex=16 pey=8 nx=16 ny=16 nz=10 kba=10 fields_per_cell=10 iterations=1 computetime=1
[MOTIF] Allreduce
[MOTIF] Fini

feldergast, 2020-12-17:
Sorry for all the bold text in the last post.  Apparently the "#" symbol gives you bold in these posts.  They were supposed to be commented lines.

DanieleDeSensi, 2021-01-30:
Thanks! That was very clear.