sstsimulator/sst-elements #1510: can someone please explain the different "arielmod"  : "1" and "0"?

RomarioNan, 2020-07-15:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

2 - Describe how to reproduce

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2020-07-15:
Ariel can optionally delay starting simulation until the simulated application gets to a certain point in the application. To do this, call "ariel_enable()" in the simulated application when you want Ariel to start simulating (the sst-tools repo provides a library with this call). Ariel mode 0 means that Ariel will start simulating the application from the beginning. Ariel mode 1 means that Ariel will wait until it encounters an "ariel_enable()" function call in the application to start simulating.

RomarioNan, 2020-07-15:
what if I want to capture all the memory access and write the trace out? which mode should I use?

gvoskuilen, 2020-07-15:
In that case, you want these parameters for ariel:
"arielmode": 0,
"tracegen": "ariel.TextTraceGenerator" # or 'ariel.CompressedBinaryTraceGenerator'

This both simulates the application and generates per-core traces named ariel-core-*. You can control the "ariel-core-" prefix by also setting the parameter "tracer.trace_prefix" in Ariel.

RomarioNan, 2020-07-22:
Thanks for your help

closed 2020-07-22 by RomarioNan