sstsimulator/sst-elements #1521: Issues when sending multiple write and read requests using the simpleMem interface to the memory hierarchy.

ainsleyrutterford, 2020-08-11:
I'm attempting to use SST to model a memory hierarchy to use with a single core CPU simulator. The CPU performs some initial set up before the clock starts ticking (such as initialising certain memory addresses that correspond to the stack pointer etc.).

I can't figure out how to access memory through any means other than the `simpleMem` `sendRequest()` function, so for now I am simply attempting to iterate through the CPU memory and send a write request per address. The issue is that when I attempt this, different crashes occur depending on the size of the memory that is being copied.

I have created a minimal working example available in [this repository](https://github.com/ainsleyrutterford/sst-mwe). It is a simple component that first initialises an array with arbitrary values, then begins the simulation and sends one write request per tick. The write requests are all of size `1` and are to each address in ascending order. These write requests are storing the contents of the array to the SST memory hierarchy. Once all of the array has been stored in the SST memory hierarchy, the component then performs one read request per tick to each memory address to check if the stored values are correct.

When running this component with an array of varying sizes, the simulation crashes or fails in different ways:

```
memorySize < 2974          - runs fine, but some addresses between 2064 and 2078 are wrong
2974 <= memorySize < 6165  - crashes after simulation is complete
6165 <= memorySize < 7553  - segfaults while checking memory
7553 <= memorySize         - segfaults while completing initial memory transfer
```

Output when running with `memorySize = 2973`:

```
(base) bash-4.2$ sst test.py 
testElement-> Using memory of size 2973
testElement-> CPU clock configured for 100MHz
testElement-> Loading memory interface: memHierarchy.memInterface ...
testElement-> Memory interface memHierarchy.memInterface successfully loaded.
testElement-> Component setup successfully.
testElement-> Transferring memory...
testElement-> Memory transferred.
testElement-> Checking memory is correct...
testElement-> Mismatch at address: 2064. Values: 135 and 32.
testElement-> Mismatch at address: 2066. Values: 0 and 34.
testElement-> Mismatch at address: 2068. Values: 0 and 36.
testElement-> Mismatch at address: 2070. Values: 0 and 38.
testElement-> Mismatch at address: 2072. Values: 136 and 40.
testElement-> Mismatch at address: 2074. Values: 0 and 42.
testElement-> Mismatch at address: 2076. Values: 0 and 44.
testElement-> Mismatch at address: 2078. Values: 0 and 46.
testElement-> Memory checks complete.
testElement-> Component finished.
Simulation is complete, simulated time: 59.47 us
```

Output when running with `memorySize = 2974`:
```
(base) bash-4.2$ sst test.py 
testElement-> Using memory of size 2974
testElement-> CPU clock configured for 100MHz
testElement-> Loading memory interface: memHierarchy.memInterface ...
testElement-> Memory interface memHierarchy.memInterface successfully loaded.
testElement-> Component setup successfully.
testElement-> Transferring memory...
testElement-> Memory transferred.
testElement-> Checking memory is correct...
testElement-> Mismatch at address: 2064. Values: 136 and 32.
testElement-> Mismatch at address: 2066. Values: 0 and 34.
testElement-> Mismatch at address: 2068. Values: 0 and 36.
testElement-> Mismatch at address: 2070. Values: 0 and 38.
testElement-> Mismatch at address: 2072. Values: 137 and 40.
testElement-> Mismatch at address: 2074. Values: 0 and 42.
testElement-> Mismatch at address: 2076. Values: 0 and 44.
testElement-> Mismatch at address: 2078. Values: 0 and 46.
testElement-> Memory checks complete.
testElement-> Component finished.
Simulation is complete, simulated time: 59.49 us
*** Error in `sst': free(): invalid next size (fast): 0x0000000000892be0 ***
======= Backtrace: =========
/lib64/libc.so.6(+0x816b9)[0x2aaaac60d6b9]
sst(_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_PN3SST3ELI17ProvidesInterfaceEESt10_Select1stISC_ESt4lessIS5_ESaISC_EE8_M_eraseEPSt13_Rb_tree_nodeISC_E+0x31)[0x49f831]
sst(_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_PN3SST3ELI17ProvidesInterfaceEESt10_Select1stISC_ESt4lessIS5_ESaISC_EE8_M_eraseEPSt13_Rb_tree_nodeISC_E+0x1b)[0x49f81b]
sst(_ZNSt8_Rb_treeINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESt4pairIKS5_PN3SST3ELI17ProvidesInterfaceEESt10_Select1stISC_ESt4lessIS5_ESaISC_EE8_M_eraseEPSt13_Rb_tree_nodeISC_E+0x1b)[0x49f81b]
sst(_ZNSt10unique_ptrISt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEES0_IS6_PN3SST3ELI17ProvidesInterfaceESt4lessIS6_ESaISt4pairIKS6_SA_EEESC_SaISD_ISE_SH_EEESt14default_deleteISK_EED1Ev+0x4d)[0x49f98d]
/lib64/libc.so.6(+0x39cd9)[0x2aaaac5c5cd9]
/lib64/libc.so.6(+0x39d27)[0x2aaaac5c5d27]
/lib64/libc.so.6(__libc_start_main+0xfc)[0x2aaaac5ae54c]
sst[0x492d29]
======= Memory map: ========
00400000-0059e000 r-xp 00000000 aff:8924c 162130025460540895             /lustre/home/br-arutterford/sst/local/sstcore-10.0.0/libexec/sstsim.x
0079d000-0079e000 r--p 0019d000 aff:8924c 162130025460540895             /lustre/home/br-arutterford/sst/local/sstcore-10.0.0/libexec/sstsim.x
0079e000-007a0000 rw-p 0019e000 aff:8924c 162130025460540895             /lustre/home/br-arutterford/sst/local/sstcore-10.0.0/libexec/sstsim.x
007a0000-00c11000 rw-p 00000000 00:00
...
Aborted
```

Output when running with `memorySize = 6165`:
```
(base) bash-4.2$ sst test.py 
testElement-> Using memory of size 6165
testElement-> CPU clock configured for 100MHz
testElement-> Loading memory interface: memHierarchy.memInterface ...
testElement-> Memory interface memHierarchy.memInterface successfully loaded.
testElement-> Component setup successfully.
testElement-> Transferring memory...
testElement-> Memory transferred.
testElement-> Checking memory is correct...
Segmentation fault
```

Output when running with `memorySize = 7553`:
```
(base) bash-4.2$ sst test.py 
testElement-> Using memory of size 7553
testElement-> CPU clock configured for 100MHz
testElement-> Loading memory interface: memHierarchy.memInterface ...
testElement-> Memory interface memHierarchy.memInterface successfully loaded.
testElement-> Component setup successfully.
testElement-> Transferring memory...
Segmentation fault
```

The parameters used for the memory hierarchy components and the links between them are shown in [test.py](https://github.com/ainsleyrutterford/sst-mwe/blob/master/test.py). They are basically copied from [exercise 1](https://github.com/sstsimulator/sst-tutorials/blob/master/exercises/ex1/ex1.py) from the sst-tutorials repo.

I don't think I'm missing anything, I should be able to write to SST memory via the `simpleMem` interface one address at a time without it crashing right? The requests are of size `1` so no misaligned loads should be being requested which I initially thought may be have been the issue due to [#700](https://github.com/sstsimulator/sst-elements/issues/700). Also, why have some addresses between 2064 and 2078 been stored incorrectly?

### Operating system and SST versions:

**Operating System:** Red Hat Enterprise Linux Server release 7.7

**SST-Core Version 10.0.0** configured with the `--disable-mpi` option and compiled with GCC 8.1.0.

**SST-Elements Version 10.0.0** compiled with GCC 8.1.0.

nmhamster, 2020-08-11:
Hi @ainsleyrutterford - just trying to work out what you want to do here, when the CPU is configured (i.e. prior to the actual simulation starting) you'd like to configure a fairly large chunk of the memory space (loading in the binary, config etc)? You are allowed to send data into the memory subsystem prior to the simulation starting to configure it. I typically create a large payload (these can be tens - hundreds of MB) starting at address zero and then putting the entire contents into the payload so it is cleanly laid out in memory. Is that a good summary of the context?

nmhamster, 2020-08-11:
I found some very simple code in the Juno CPU example of how you want to push data to the memory system (during initialization, not during a run). The example below loads in an "executable" and pushes this into the memory system using the `sendInitData` capability that can only be used during SST's `init` phases. After these are completed, only standard `sendRequest` operations can be completed.

```
void JunoCPU::init( unsigned int phase ) {
    mem->init( phase );

    if( 0 == phase ) {
        const size_t initLen = static_cast<size_t>( progReader->getDataLength() + progReader->getInstLength() );

        std::vector<uint8_t> exeImage;
        exeImage.reserve( initLen );

        for( size_t i = 0; i < initLen; ++i ) {
            exeImage.push_back( progReader->getBinaryBuffer()[i] );
        }

	for( size_t i = 0; i < progReader->getPadding(); ++i ) {
	    exeImage.push_back( static_cast<uint8_t>(0) );
	}

        SimpleMem::Request* writeExe = new SimpleMem::Request(SimpleMem::Request::Write, 0, exeImage.size(), exeImage);
        output.verbose(CALL_INFO, 1, 0, "Sending initialization data to memory...\n");

        mem->sendInitData(writeExe);

        output.verbose(CALL_INFO, 1, 0, "Initialization data sent.\n");
    }
}
```

ainsleyrutterford, 2020-08-11:
Hi @nmhamster, thanks for the quick reply! Yep that's correct that's exactly what I'd like to do, I'll try implementing that now. Before using `sendInitData` is it necessary to call `mem->init()` like in the example above?

nmhamster, 2020-08-11:
@ainsleyrutterford - yes, I believe so. This permits the memory system to discover its hierarchy etc.

ainsleyrutterford, 2020-08-11:
I've implemented a simple example using `sendInitData` and I can't seem to get it working correctly. When I read values back from the memory hierarchy they are all `0` apparently. The [minimal working example](https://github.com/ainsleyrutterford/sst-mwe) has been updated if you want to take a look but I'll paste my `init` function here as well:

```c++
void testElement::init(unsigned int phase) {
  // Initialise the memory array with some arbitrary values between 0 and 126
  memory = new uint8_t[memorySize]();
  for (int i = 0; i < memorySize; i++) {
    memory[i] = i % 127;
  }
  // Initialise variables
  address = 0;
  finished = false;

  mem->init(phase);

  if (phase == 0) {
    const size_t initLen = static_cast<size_t>(memorySize);
    std::vector<uint8_t> image;
    image.reserve(memorySize);

    for (size_t i = 0; i < memorySize; i++) {
      image.push_back(memory[i]);
    }

    SimpleMem::Request* req = new SimpleMem::Request(SimpleMem::Request::Write,
                                                     0, image.size(), image);
    output.verbose(CALL_INFO, 1, 0, "Transferring data to memory...\n");
    mem->sendInitData(req);
    output.verbose(CALL_INFO, 1, 0, "Memory transferred.\n");
  }
}
```

This is the output I get when running the MWE:
```
(base) bash-4.2$ sst test.py 
testElement-> Using memory of size 10
testElement-> CPU clock configured for 100MHz
testElement-> Loading memory interface: memHierarchy.memInterface ...
testElement-> Memory interface memHierarchy.memInterface successfully loaded.
testElement-> Transferring data to memory...
testElement-> Memory transferred.
testElement-> Mismatch at address: 1. Values: 0 and 1.
testElement-> Mismatch at address: 2. Values: 0 and 2.
testElement-> Mismatch at address: 3. Values: 0 and 3.
testElement-> Mismatch at address: 4. Values: 0 and 4.
testElement-> Mismatch at address: 5. Values: 0 and 5.
testElement-> Mismatch at address: 6. Values: 0 and 6.
testElement-> Mismatch at address: 7. Values: 0 and 7.
testElement-> Mismatch at address: 8. Values: 0 and 8.
testElement-> Mismatch at address: 9. Values: 0 and 9.
testElement-> Memory checks complete.
testElement-> Component finished.
Simulation is complete, simulated time: 110 ns
```

Can you see anything obvious I'm doing wrong?

nmhamster, 2020-08-11:
@ainsleyrutterford - I think you have the `backing` set to `none` which effectively tells the memory to not respond to read/write requests with payload (it times correctly, but it does not save the data). Can you just try removing that setting (Line 41 in your `test.py`)?

ainsleyrutterford, 2020-08-11:
@nmhamster that did it! I had `"do_not_back" : 1` left over from [exercise 1](https://github.com/sstsimulator/sst-tutorials/blob/master/exercises/ex1/ex1.py). A warning then told me to: `Use 'backing' parameter instead and specify 'none', 'malloc', or 'mmap'`. Not sure why I chose `none`! Thank you very much I'll close this now as it seems to be working great.

closed 2020-08-11 by ainsleyrutterford

nmhamster, 2020-08-11:
@ainsleyrutterford - thanks for letting us know. Give us a shout if you have any additional questions.