sstsimulator/sst-elements #1535: Ariel runstream examples doesn't print CacheHit/Miss stats

shubhamn94, 2020-09-18:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement - 

> I am running Ariel runstream examples (runstream.py, runstreamSt.py, runstreamNB.py). These script contains statistics for L1 Cache Hits and Misses. However, when I execute runstream.py it doesn't print CacheHit/Miss stats. Though, other Ariel related stats are printed.
I looked into the cacheController.h file in memHierarchy but I don't see any knob in SST_ELI_DOCUMENT_STATISTICS to report _'CacheHits'_ and _'CacheMisses'_.  So is it the issue with the python file? 
Also, am I looking at the correct file (cacheController.h) for these stats or is it that I am missing something?  
Below are the lines from runstream.py for printing cache stats.
```
ariel.enableStatistics([
      "cycles",
      "active_cycles",
      "instruction_count",
      "read_requests",
      "write_requests"
])

l1cache.enableStatistics([
      "CacheHits",
      "CacheMisses"
])
```
Output:

```
SSTARIEL: Function profiling is disabled.
ARIEL-SST: Did not find ARIEL_OVERRIDE_POOL in the environment, no override applies.
ARIEL-SST PIN tool activating with 1 threads
ARIEL: Default memory pool set to 0
ARIEL: Tool is configured to begin with profiling immediately.
ARIEL: Starting program.
Identified routine: clock_gettime, replacing with Ariel equivalent...
Replacement complete.
Identified routine: clock_gettime, replacing with Ariel equivalent...
Replacement complete.
Allocating arrays of size 2000 elements.
Done allocating arrays.
Perfoming the fast_c compute loop...
Sum of arrays is: 6999500.000000
Freeing arrays...
Done.
CORE ID: 0 PROCESSED AN EXIT EVENT

Ariel Memory Management Statistics:
---------------------------------------------------------------------
Page Table Sizes:
- Map entries         140
Page Table Coverages:
- Bytes               573440
 a0.read_requests.0 : Accumulator : Sum.u64 = 90989; SumSQ.u64 = 90989; Count.u64 = 90989; Min.u64 = 1; Max.u64 = 1; 
 a0.write_requests.0 : Accumulator : Sum.u64 = 42466; SumSQ.u64 = 42466; Count.u64 = 42466; Min.u64 = 1; Max.u64 = 1; 
 a0.instruction_count.0 : Accumulator : Sum.u64 = 401500; SumSQ.u64 = 401500; Count.u64 = 401500; Min.u64 = 1; Max.u64 = 1; 
 a0.cycles.0 : Accumulator : Sum.u64 = 12115798; SumSQ.u64 = 12115798; Count.u64 = 12115798; Min.u64 = 1; Max.u64 = 1; 
 a0.active_cycles.0 : Accumulator : Sum.u64 = 304923; SumSQ.u64 = 304923; Count.u64 = 304923; Min.u64 = 1; Max.u64 = 1; 
Simulation is complete, simulated time: 12.1158 ms

```
2 - Describe how to reproduce
`sst <SOURCE_DIR>/sst-elements/src/sst/elements/ariel/frontend/simple/examples/stream/runstream.py`

3 - What Operating system(s) and versions 
```
Ubuntu 18.04.4 LTS (Bionic Beaver)
Linux kernel version - Linux 5.4.0-45-generic x86_64
GCC version - 4.8.4
```
4 - What version of external libraries (Boost, MPI)
```
PIN version - pin-3.13-98189-g60a6ef199-gcc-linux
Boost - N/A
MPI version - 4.0.5

```
5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
```
sst-core --> commit e60bce8dde2e4e1e57ecc0e7765f46a8c5e1d63b (HEAD -> master, origin/master)
sst-elements --> commit 1e3ba3039b6ae64721e515b42258acb572497b0a (HEAD -> master, origin/master)
```

6 - Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2020-09-18:
Thanks for pointing this out, it looks like the cache hit/miss summary stats aren't in the statistics. I've pushed a change to add them but it will be a day or two before they're available in the master branch. In the meantime, there are some more fine-grained statistics you can enable (GetSHit_Arrival, GetSHit_Blocked, GetSMiss_Arrival, GetSMiss_Blocked, GetXHit_Arrival, GetXHit_Blocked, GetXMiss_Arrival, GetXMiss_Blocked), getting the total hits/misses requires summing the *Miss* and *Hit* stats.

shubhamn94, 2020-09-21:
Thank you for resolving it.

closed 2020-09-21 by shubhamn94