sstsimulator/sst-elements #1549: Messier: Memory size set doesn't seem to be passed

renxida, 2020-10-19:
# Overview

### What I think is the issue:
I'm suspecting that in python test scripts `nvm_memory.setSubComponent("backend", "memHierarchy.Messier")` isn't passing memory size properly to the backend.

### What I did:
1. sets up 1MB of memory in Messier
2. uses Miranda's copy generator to copy 128 MB of memory from 0-128MB to 128-256MB, touching 256 MB in total.

### What happened:
simulation finished without problem.

### What I expect:
failure because I'm attempting to touch more memory than I set up.

# How to Reproduce the issue

### SST Version

##### Core
https://github.com/sstsimulator/sst-core/releases/tag/v10.0.0_Final

##### Elements
I am on sst-elements release 10.0.0

https://github.com/sstsimulator/sst-elements/releases/tag/v10.0.0_Final

SST-Elements 10.0.0 Release
@allevin allevin released this on May 19

SHA1 = 25a40398c48abb47032840107d757c8dc48ab3a5

### OS Version
I was running it on Lawrence Berkeley's Banner host, `uname -a` output bleow:
```Linux banner 4.15.0-121-generic #123-Ubuntu SMP Mon Oct 5 16:16:40 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux```

But the same issue happens on my personal computer.

### Boost version
`Version: 1.65.1.0ubuntu1`

### Open MPI version:
https://termbin.com/3kvf

I ran a modification of https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/Messier/tests/gupsgen.py

```python
import sys
import argparse
print("entering...")
parser = argparse.ArgumentParser()
parser.add_argument("--memory_mb", default=1, type=int)
parser.add_argument("--tocopy_mb", default=128, type=int) # should 
args = parser.parse_args()

print("Arguments: ", args)

import sst

# Define SST core options
sst.setProgramOption("timebase", "1ps")
sst.setProgramOption("stopAtCycle", "0 ns")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

# Define the simulation components
comp_cpu = sst.Component("cpu", "miranda.BaseCPU")
comp_cpu.addParams({
    "verbose" : 0,
})
# cpugen = comp_cpu.setSubComponent("generator", "miranda.GUPSGenerator")
# cpugen.addParams({
#     "verbose" : 0,
#     "startat": 0,
#     "count": args.count,
#     "max_address" : (args.memory_mb) * 1024 * 1024,
# })

cpugen = comp_cpu.setSubComponent("generator", "miranda.CopyGenerator")
cpugen.addParams({
    "verbose" : 1,
    "operandwidth": 8, # 8 bytes per copy
    "request_count": args.tocopy_mb * 1024 * 1024 / 8 # mb divided by bits per request. 
})

# Enable statistics outputs
comp_cpu.enableAllStatistics({"type":"sst.AccumulatorStatistic"})

comp_l1cache = sst.Component("l1cache", "memHierarchy.Cache")
comp_l1cache.addParams({
      "access_latency_cycles" : "2",
      "cache_frequency" : "2 Ghz",
      "replacement_policy" : "lru",
      "coherence_protocol" : "MESI",
      "associativity" : "4",
      "cache_line_size" : "64",
      "prefetcher" : "cassini.StridePrefetcher",
      "L1" : "1",
      "cache_size" : "8KB",
      "backing" : "none",
})

# Enable statistics outputs
comp_l1cache.enableAllStatistics({"type":"sst.AccumulatorStatistic"})

nvm_memory = sst.Component("memory", "memHierarchy.MemController")
nvm_memory_backend = nvm_memory.setSubComponent("backend", "memHierarchy.Messier")

nvm_mem_params = {
    "clock" : "1024 MHz",
    "backing" : "none",
}
nvm_backend_params = {
   # "max_requests_per_cycle" : 1,
    "mem_size" : f"{args.memory_mb}MiB", 
    #"backendConvertor.backend.clock" : "1024 MHz",
    #"backendConvertor" : "memHierarchy.MemBackendConvertor", 
   # "backend.device_count" : 1,
   # "backend.link_count" : 4,
   # "backend.vault_count" : 16,
   # "backend.queue_depth" : 64,
   # "backend.bank_count" : 16,
   # "backend.dram_count" : 20,
   # "backend.capacity_per_device" : 4, # Min is now 4 but we'll just use 1 of it
   # "backend.xbar_depth" : 128,
   # "backend.max_req_size" : 64,
   # "backend.tag_count" : 512,
}

nvm_memory.addParams(nvm_mem_params)
nvm_memory_backend.addParams(nvm_backend_params)

messier_inst = sst.Component("NVMmemory", "Messier")

messier_params = {
	"clock" : "1 GHz",

}
messier_inst.addParams(messier_params)

messier_inst.addParams({
      "tCL" : "30",
      "tRCD" : "300",
      "clock" : "1GHz",
      "tCL_W" : "1000",
      "write_buffer_size" : "32",
      "flush_th" : "90",
      "num_banks" : "32",
      "max_outstanding" : "32",
      "max_current_weight" : "160",
      "read_weight" : "5",
      "write_weight" : "50",
      "max_writes" : 4
})

#nvm_memory.addParams({
 #     "coherence_protocol" : "MESI",
#      "backend.access_time" : "1000 ns",
 #     "backend.mem_size" : str(memory_mb * 1024 * 1024) + "B",
#      "clock" : "1GHz"
#})

#nvm_memory.addParams(nvm_mem_params)

link_nvm_bus_link = sst.Link("link_nvm_bus_link")
link_nvm_bus_link.connect( (messier_inst, "bus", "50ps"), (nvm_memory_backend, "nvm_link", "50ps") )

# Define the simulation links
link_cpu_cache_link = sst.Link("link_cpu_cache_link")
link_cpu_cache_link.connect( (comp_cpu, "cache_link", "1000ps"), (comp_l1cache, "high_network_0", "1000ps") )
link_cpu_cache_link.setNoCut()

link_mem_bus_link = sst.Link("link_mem_bus_link")
link_mem_bus_link.connect( (comp_l1cache, "low_network_0", "50ps"), (nvm_memory, "direct_link", "50ps") )
```

renxida, 2020-10-19:
I realize that this might be due to Messier gracefully handling out-of-bounds memory accesses, and I would really appreciate it if somebody could point me to where that happens. I'm rather new to SST development.

gvoskuilen, 2020-10-19:
It looks like this is actually a bug - the memory controller is not enforcing the mem_size parameter and it should be. I've created a fix for this. Thanks for raising the issue.

renxida, 2020-10-27:
@gvoskuilen hey! thanks for fixing this! I really appreciate the responsiveness.

Just so I understand what I'm doing... Does this mean that in my script I failed to set a size for the underlying Messier NVM backend, and I'm only setting a bound for the memory controller, which is not enforced?

gvoskuilen, 2020-10-29:
That is correct. The "memory controller" component only acts as an interface to a memory model (which sometimes includes an actual controller model) and it needs to know the memory size so that it can enforce bounds. To limit Messier itself, you need to set Messier's memory size parameter. Run 'sst-info Messier' to see Messier's available parameters.

renxida, 2020-10-30:
Gocha! Thank you!

closed 2021-07-15 by hughes-c