sstsimulator/sst-elements #1558: SST Elements compile with '-g' but not '-g3' or '-ggdb'

renxida, 2020-11-03:
SST-elements doesn't compile with '-g3' option to produce extra debug info.
Make complains of "3: file not found"

Similarly, '-ggdb' doesn't work.
with "file gdb not found"

However, '-g' works.

I grepped for '-g3' in the makefiles generated by ./configure , and turns out 3 is being split from '-g3' and treated as a separate file:

```src/sst/elements/CramSim/Makefile:287:CXXFLAGS = -g3 -O0 3 -std=c++1y -D__STDC_FORMAT_MACROS```

The script i used to run compilation is as follows:

```
set -e # exit on command failure
PYCONFIG=/usr/bin/python3-config
SSTCOREPREFIX=$HOME/local/sst/sstcore-10.0.0
SSTELEMENTSPREFIX=$HOME/local/sst/sstelements-10.0.0
DEBUG_FLAGS="CPPFLAGS='-DDEBUG' CXXFLAGS='-DDEBUG' CFLAGS='-g3 -O0' CXXFLAGS='-g3 -O0'"
cd sstcore-10.0.0
ls
eval $DEBUG_FLAGS ./configure --prefix=$SSTCOREPREFIX --disable-mpi --with-python=$PYCONFIG
make -j4
make install
cd ..
cd sst-elements-library-10.0.0
eval $DEBUG_FLAGS ./configure --prefix=$SSTELEMENTSPREFIX --disable-mpi --with-sst-core=$SSTCOREPREFIX --with-python=$PYCONFIG
make -j4
make install
```

hughes-c, 2021-11-19:
This is still a problem.

sst-core:
`CFLAGS="-ggdb3" CXXFLAGS="-ggdb3" ./configure
                  C Compiler Options : -ggdb3 -Wall -Wextra
                  C++ Compiler Options : -ggdb3 -std=c++1y -D__STDC_FORMAT_MACROS -Wall -Wextra`

sst-elements:
`                    C Compiler Options : -g -O2 gdb3 
                     C++ Compiler Options : -g -O2 gdb3 -std=c++1y -D__STDC_FORMAT_MACROS `

berquist, 2023-07-17:
The problem is with `sst-config`:
```
$ ../install/bin/sst-config --CFLAGS
3
```
and the actual contents of `sst_config.h` are ok:
```c
/* Defines the CFLAGS used to build SST */
#define SST_CFLAGS "-g3 -Wall -Wextra"
```

berquist, 2023-07-17:
I misunderstood, it's from `sstsimulator.conf`, and the problem is in `configure.ac`:
```autoconf
dnl Remove flags like -g, -O, and -W, that are not needed by other elements
SST_EXPORT_CXXFLAGS=`echo "$CXXFLAGS" | sed -E -e 's/ *-(g|O@<:@^@<:@:space:@:>@@:>@*|W@<:@^@<:@:space:@:>@@:>@+)//g'`
SST_EXPORT_CFLAGS=`echo "$CFLAGS" | sed -E -e 's/ *-(g|O@<:@^@<:@:space:@:>@@:>@*|W@<:@^@<:@:space:@:>@@:>@+)//g'`
```

hughes-c, 2023-08-14:
I think this was fixed in https://github.com/sstsimulator/sst-elements/commit/aac0adcbb38433519b87fb0e122d824581e223d5 for elements.

berquist, 2023-08-14:
Yes, but please test it for me.

berquist, 2023-08-23:
Yep, tried again with both `-g3` and `-ggdb`.  Fixed by https://github.com/sstsimulator/sst-core/pull/974.

closed 2023-08-23 by berquist