sstsimulator/sst-elements #1567: Simulation terminates quickly for 1 core on memory but runs indefinitely for 4 cores

renxida, 2020-12-08:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

Using multiple cores accessing the same messier memory, when running on 4 cores instead of 1, experiment never terminates.

2 - Describe how to reproduce

I'm running this SST config script: https://gist.github.com/renxida/f34ca133a4ddf6a126e331ce10cb6145.

3 - What Operating system(s) and versions 

```
No LSB modules are available.
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic
```

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

10.0.0 unmodified

6 - Fill out Labels, Milestones, and Assignee fields as best possible

renxida, 2020-12-08:
Oh I don't know if this is relevent, but I'm also using valgrind massif to trace the simulation.

allevin, 2020-12-08:
@gvoskuilen Not sure if you are the one to look this over...

nmhamster, 2021-01-12:
@renxida - can you try running with the heartbeat function turned on in the core (add `--heartbeat-period=10ns` or something similar). This will have SST print out heart beat information to check it is progressing time during execution. This won't fix the bug but allows us to check that the simulation has not stalled somewhere.

hughes-c, 2023-06-22:
No follow-up; closing.

closed 2023-06-22 by hughes-c