sstsimulator/sst-elements #1613: GOBLIN HMCSIM integration does not produce accurate results, Please help.

Shubhang-Pandey, 2021-03-03:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

2 - Describe how to reproduce

3 - What Operating system(s) and versions 

4 - What version of external libraries (Boost, MPI)

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible

gvoskuilen, 2021-03-09:
The HMCSim simulator/integration is maintained by a different group: https://github.com/tactcomplabs/gc64-hmcsim

If the inaccuracy is specific to the HMCSim simulator integration, please contact the HMCSim developers. However, if the issue is located in SST (not due to the HMCSim backend), can you provide additional details about what is inaccurate?

Shubhang-Pandey, 2021-03-10:
It's hard to tell exactly on which end the inaccuracy exists. I have tried to issue packets of different sizes and the results are very inconsistent. Offloading atomics to HMC, results do not appear.

closed 2021-04-27 by Shubhang-Pandey