sstsimulator/sst-elements #1616: Documentation Bug for Deprecated Components - Incorrect List

nmhamster, 2021-03-10:
We had an email from a user who said the deprecated list of components here is not correct - Vanadis is listed which it should not be. This has caused confusion.

URL: http://sst-simulator.org/SSTPages/SSTmicroReleaseV10dot0dot0/

allevin, 2021-03-11:
@nmhamster I figured this one out... 

Per Issue #10 we moved the vanadis element into the element archive around Jan 29 2020.  At the time we moved it, the only thing in the vanadis directory was an Makefile.am file.  Prior to that, Vanadis was not being distributed in our releases, but was in the devel and master branches.   From what I can see, the old (empty) Vanadis was not included in any release.  

The release notes on 10.0 was incorrect... Vanadis was not deprecated... It was just removed from the devel branch and moved to the archive.  

Since then, Vanadis has raised from the ashes (like a Phoenix) and will be released as a new element.  

Do you want me to modify the 10.0 release notes and remove the comment on Vanadis?   Its a quick and easy change...

nmhamster, 2021-03-11:
Hi @allevin - can we modify the 10.0 release notes to remove Vanadis from the deprecated list please? Thank you.

allevin, 2021-03-12:
@nmhamster - Fixed the SST 10.0 Release document in the website.  Also dbl-checked the README.md in sst-elements and it only refers to the online document.
http://sst-simulator.org/SSTPages/SSTmicroReleaseV10dot0dot0/

closed 2021-03-12 by allevin

nmhamster, 2021-03-12:
@allevin - thanks for the fix. Looks good.