sstsimulator/sst-elements #1655: SST Core: Error:  TimeConverter creation requires a time unit (s or Hz), 0 b/s was passed to call

Shubhang-Pandey, 2021-04-27:
Please help with this error. I am trying for network simulations and with the ariel processor model.

feldergast, 2021-04-27:
This typically means that you put in a bandwidth to a parameter that is looking for a frequency or period.  I'd need to see your configuration file to provide any deeper insight.

Shubhang-Pandey, 2021-04-28:
Thank you for the support. Please have a look at the script.

import sst
import os

next_core_id = 0
next_network_id = 0
next_memory_ctrl_id = 0

clock = "2660MHz"
memory_clock = "200MHz"
coherence_protocol = "MESI"

cores_per_group = 2
active_cores_per_group = 2
memory_controllers_per_group = 1
groups = 4
os.environ["OMP_NUM_THREADS"]=str(groups * cores_per_group)

l3cache_blocks_per_group = 5
l3cache_block_size = "1MB"

ring_latency = "50ps"
ring_bandwidth = "85GB/s"
ring_flit_size = "72B"

memory_network_bandwidth = "85GB/s"

mem_interleave_size = 4096  # Do 4K page level interleaving
memory_capacity = 16384     # Size of memory in MBs

streamN = 1000000

l1_prefetch_params = {
        }

l2_prefetch_params = {
    "prefetcher": "cassini.StridePrefetcher",
    "reach": 16,
    "detect_range" : 1
}

ringstop_params = {
    #"topoHyperX:shape" : "4x4",
	"topology" : "merlin.dragonfly",
    "output_latency" : "100ps",
    "xbar_bw" : ring_bandwidth,
    "input_buf_size" : "2KB",
    "input_latency" : "100ps",
    "num_ports" : "3",
    "debug" : "0",
#    "torus:local_ports" : "1",
    "flit_size" : ring_flit_size,
    "output_buf_size" : "2KB",
    "link_bw" : ring_bandwidth,
#   "torus:width" : "1",
#   "topology" : "merlin.torus"
    #"checkerboard :  "1"
    "xbar_arb" :  "merlin.xbar_arb_lru",
	"dragonfly:hosts_per_router" : "2",
    "dragonfly:routers_per_group" : cores_per_group,
    "dragonfly:intergroup_links" : "1",
    "dragonfly:num_groups" : groups,
    "dragonfly:algorithm" : "minimal",
}

topology_params = {
    "hosts_per_router" : "2",
    "routers_per_group" : cores_per_group,
    "intergroup_links" : "1",
[... 192 lines cut ...]
            "memNIC.addr_range_start" : next_memory_ctrl_id * mem_interleave_size,
            "memNIC.addr_range_end" : (memory_capacity * 1024 * 1024) - (groups * memory_controllers_per_group * mem_interleave_size) + (next_memory_ctrl_id * mem_interleave_size)
        })
        dc.addParams(dc_params)

        memLink = sst.Link("mem_link_" + str(next_memory_ctrl_id))
        memLink.connect((mem, "direct_link", ring_latency), (dc, "memory", ring_latency))

        netLink = sst.Link("dc_link_" + str(next_memory_ctrl_id))
        netLink.connect((dc, "network", ring_latency), (router_map["rtr." + str(next_network_id)], "port2", ring_latency))

        next_network_id = next_network_id + 1
        next_memory_ctrl_id = next_memory_ctrl_id + 1

# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(4)
sst.enableAllStatisticsForAllComponents({"type":"sst.AccumulatorStatistic"})

sst.setStatisticOutput("sst.statOutputCSV")
sst.setStatisticOutputOptions( {
    "filepath"  : "./stats-snb-ariel.csv",
    "separator" : ", "
} )

print("Completed configuring the SST Sandy Bridge model")

Shubhang-Pandey, 2021-04-28:
Thank you for your support. Please find the script. 
I am a little new to the topic, for further errors if present please do tell.

I have taken the ariel-snb-mlm.py file as reference, to start with,

import sst
import os

next_core_id = 0
next_network_id = 0
next_memory_ctrl_id = 0

clock = "2660MHz"
memory_clock = "200MHz"
coherence_protocol = "MESI"

cores_per_group = 2
active_cores_per_group = 2
memory_controllers_per_group = 1
groups = 4
os.environ["OMP_NUM_THREADS"]=str(groups * cores_per_group)

l3cache_blocks_per_group = 5
l3cache_block_size = "1MB"

ring_latency = "50ps"
ring_bandwidth = "85GB/s"
ring_flit_size = "72B"

memory_network_bandwidth = "85GB/s"

mem_interleave_size = 4096  # Do 4K page level interleaving
memory_capacity = 16384     # Size of memory in MBs

streamN = 1000000

l1_prefetch_params = {
        }

l2_prefetch_params = {
    "prefetcher": "cassini.StridePrefetcher",
    "reach": 16,
    "detect_range" : 1
}

ringstop_params = {
    #"topoHyperX:shape" : "4x4",
	"topology" : "merlin.dragonfly",
    "output_latency" : "100ps",
    "xbar_bw" : ring_bandwidth,
    "input_buf_size" : "2KB",
    "input_latency" : "100ps",
    "num_ports" : "3",
    "debug" : "0",
#    "torus:local_ports" : "1",
    "flit_size" : ring_flit_size,
    "output_buf_size" : "2KB",
    "link_bw" : ring_bandwidth,
#   "torus:width" : "1",
#   "topology" : "merlin.torus"
    #"checkerboard :  "1"
    "xbar_arb" :  "merlin.xbar_arb_lru",
	"dragonfly:hosts_per_router" : "2",
    "dragonfly:routers_per_group" : cores_per_group,
    "dragonfly:intergroup_links" : "1",
    "dragonfly:num_groups" : groups,
    "dragonfly:algorithm" : "minimal",
}

topology_params = {
[... 195 lines cut ...]
            "memNIC.addr_range_start" : next_memory_ctrl_id * mem_interleave_size,
            "memNIC.addr_range_end" : (memory_capacity * 1024 * 1024) - (groups * memory_controllers_per_group * mem_interleave_size) + (next_memory_ctrl_id * mem_interleave_size)
        })
        dc.addParams(dc_params)

        memLink = sst.Link("mem_link_" + str(next_memory_ctrl_id))
        memLink.connect((mem, "direct_link", ring_latency), (dc, "memory", ring_latency))

        netLink = sst.Link("dc_link_" + str(next_memory_ctrl_id))
        netLink.connect((dc, "network", ring_latency), (router_map["rtr." + str(next_network_id)], "port2", ring_latency))

        next_network_id = next_network_id + 1
        next_memory_ctrl_id = next_memory_ctrl_id + 1

# Enable SST Statistics Outputs for this simulation
sst.setStatisticLoadLevel(4)
sst.enableAllStatisticsForAllComponents({"type":"sst.AccumulatorStatistic"})

sst.setStatisticOutput("sst.statOutputCSV")
sst.setStatisticOutputOptions( {
    "filepath"  : "./stats-snb-ariel.csv",
    "separator" : ", "
} )

print("Completed configuring the SST Sandy Bridge model")

Shubhang-Pandey, 2021-04-28:
Thank you, Sir, for your support.
 Sir can you please tell me whether we can simulate Memory Centric Network Architectures on SST. Are there some test files or methods which can help?
Response would be highly appreciated.

closed 2023-07-27 by hughes-c