sstsimulator/sst-elements #1678: Elements that need python to enable shared libraries should check for that at configure time.

calewis, 2021-06-07:
1 - Detailed description of problem or enhancement

Making elements and I get the following error
```
libpython3.9.a(import.o): relocation R_X86_64_32S against symbol `PyDict_Type' can not be used when making a shared object; recompile with -fPIC
```

2 - Describe how to reproduce

I didn't try in depth to figure out how to make this work, but I believe all you need is a python that was not built with `--enable-shared` and then use that python with core. 

I believe this could be detected sooner than the elements linking step. Which is why I made this issue.

hughes-c, 2023-06-22:
@feldergast do we care about this?

feldergast, 2023-06-22:
We haven't seen any issues with this.

closed 2023-06-22 by feldergast