sstsimulator/sst-elements #1683: Download package on sst-simulator.org does not include test output for test_simplesimulation_carwash

mgoldstein322, 2021-07-01:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

Installing and running SST from the simulator download page on sst-simulator.org does not include the file:
.../sst-elements/src/sst/elements/simpleSimulation/tests/refFiles/test_simpleSimulationCarWash.out

2 - Describe how to reproduce

Build SST using the tarballs on sst-simlulator.org.

3 - What Operating system(s) and versions 

Ubuntu 20.04

4 - What version of external libraries (Boost, MPI)

MPI 3.3.2
Python 3.8.10
Autoconf 2.69
Automake 1.16.1

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

core: a1d0dd4f6b0c4216c35589179da0f677bbe5fdf3
elements: 8b9e779a8ace79a2d5767692a65505c63d3c5cd1

6 - Fill out Labels, Milestones, and Assignee fields as best possible

-----
Full output:

$ sst-test-elements -w "\*simple\*"
INFO: SST Test Engine Instantiated - Running tests on Registered Elements 
INFO: Test Platform = UBUNTU 20.04
INFO: Running on Python Version = 3.8.10
INFO: TestEngine Version = (11.0.0)
INFO: Available Cores = 4; Num Ranks = 1; Num Threads = 1
INFO: TestSuite wildcards to search for: testsuite_*simple*.py
INFO: Test scenarios to filter tests: NONE
INFO: SST Test Output Directory = /home/magold/sst_test_outputs
INFO: Full colorized output can be obtained by running 'pip install blessings pygments'

\=== TESTS STARTING ===================================================
testsuite_default_simpleComponents
    PASS -- test_basic_clocks (testsuite_default_simpleComponents.testcase_simpleComponents) [0.129s]
    PASS -- test_basic_links (testsuite_default_simpleComponents.testcase_simpleComponents) [0.130s]
    PASS -- test_basic_params (testsuite_default_simpleComponents.testcase_simpleComponents) [0.130s]
    PASS -- test_basic_statistics_0 (testsuite_default_simpleComponents.testcase_simpleComponents) [0.125s]
    PASS -- test_basic_statistics_1 (testsuite_default_simpleComponents.testcase_simpleComponents) [0.125s]
    PASS -- test_basic_statistics_2 (testsuite_default_simpleComponents.testcase_simpleComponents) [0.121s]
    PASS -- test_example_0 (testsuite_default_simpleComponents.testcase_simpleComponents) [0.131s]
    PASS -- test_example_1 (testsuite_default_simpleComponents.testcase_simpleComponents) [0.126s]
testsuite_default_simpleSimulation

ERROR: Cannot diff files: Ref File /home/magold/Documents/sst/sst-elements/src/sst/elements/simpleSimulation/tests/refFiles/test_simpleSimulationCarWash.out does not exist
    FAIL -- test_simplesimulation_carwash (testsuite_default_simpleSimulation.testcase_simpleSimulation) [0.110s]

\======================================================================
\=== TESTS FINISHED ===================================================
\======================================================================

\======================================================================
FAIL: test_simplesimulation_carwash (testsuite_default_simpleSimulation.testcase_simpleSimulation)
\----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/magold/Documents/sst/sst-elements/src/sst/elements/simpleSimulation/tests/testsuite_default_simpleSimulation.py", line 20, in test_simplesimulation_carwash
    self.simplesimulation_template("simpleSimulationCarWash")
  File "/home/magold/Documents/sst/sst-elements/src/sst/elements/simpleSimulation/tests/testsuite_default_simpleSimulation.py", line 57, in simplesimulation_template
    self.assertTrue(cmp_result, "Sorted Output file {0} does not match sorted Reference File {1}".format(outfile, reffile))
AssertionError: False is not true : Sorted Output file /home/magold/sst_test_outputs/run_data/test_simpleSimulationCarWash.out does not match sorted Reference File /home/magold/Documents/sst/sst-elements/src/sst/elements/simpleSimulation/tests/refFiles/test_simpleSimulationCarWash.out

\----------------------------------------------------------------------
Ran 9 tests in 1.146s

FAILED (failures=1)

\=== TEST RESULTS BREAKDOWN ===========================================

testsuite_default_simpleSimulation.testcase_simpleSimulation
 - FAILED  : test_simplesimulation_carwash (testsuite_default_simpleSimulation.testcase_simpleSimulation)

\=== TEST RESULTS SUMMARY =============================================

Tests Run            = 9
\----------------------------------------
Passing              = 8
Failures             = 1
Skipped              = 0
Errors               = 0
Expected Failures    = 0
Unexpected Successes = 0
\----------------------------------------------------------------------
\-- Total Test Time = 0 Hours, 0 Minutes, 1.149 Seconds --

\==================================
\== TESTING FAILED DUE TO ERRORS ==
\==================================

\======================================================================

nmhamster, 2021-07-01:
@mgoldstein322 - thanks for the error report. PR #1684 should address this problem.

nmhamster, 2021-07-01:
Closing bug, please re-open if PR #1684 does not fix this issue.

closed 2021-07-01 by nmhamster