sstsimulator/sst-elements #1688: Nightly testing for make-dist on sst-elements provides a false pass if distribution files are missing some ref files

allevin, 2021-07-09:
Issue #1683 indicated that the simple carwash test was failing due to a missing test ref file.   This should have been caught by our nightly testing of the distribution.  

Analysis of the nightly build/test system for make-dist discovered that the frameworks build/test script (copied from the old bamboo script) copied the reference files from the devel branch to the extracted distribution files that were being tested.  This allowed some tests that are missing reference files from the distribution to **falsely** pass.  

Upon testing of a corrected build/test script, it was discovered that the following elements will fail testing:  
  * **simpleSimulation**  
  * **prospero**
  * **zodiac**

NOTE: This affects the SST 11.0 release.  

A new PR will soon be generated to correct the Makefile.am for the appropriate elements, and an updated frameworks build/test system will be merged.

allevin, 2021-07-09:
PR #1689 has been submitted and tests have shown correct behavior.

nmhamster, 2021-07-14:
Closing issue, can reopen if we see behavior return.

closed 2021-07-14 by nmhamster