sstsimulator/sst-elements #1692: Confusion about bus component

SteveY4ng, 2021-07-13:
Hello！

I have a simple question. But it bothers me for a long time.
I‘m confused about the[ bus component](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/bus.h) in sst-elements.
This component has a parameter, "drain_bus"([line 57](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/bus.h)), which means that you can forward all requests in one cycle. 
However, for all I know, the bus in reality can only forward one request in one cycle. But this parameter seems to tell me that there are buses that can drain requests in one cycle. I don't think it's reasonable. 
How can I understand it? Why does this parameter(drain_bus) exist？

hughes-c, 2021-07-13:
This was added during the development of shogun to emulate a system with more memory bandwidth that was supported in the memory components at the time.

nmhamster, 2021-07-13:
Just to add to what @hughes-c said, yes typically buses can only forward a single request in a cycle (that's normal behavior). This option was added to allow the models to use bus components but to push a lot more requests into the system in a cycle so we could study the effects of large numbers of events entering the memory system for a study (ignoring the bus design). These kind of studies allow us to do "what if" analyses of parts of the memory system more easily. It wouldn't used when we were developing full cycle accurate models of a system (when you would want to use the true single event per cycle nature of a bus). Hope that helps.

SteveY4ng, 2021-07-14:
It helps! Thank you guys very much! It's very kind of you guys to answer me clearly.

I have another question about the component "[memory controller](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/memoryController.h)".

I just want to use the memory controller component to construct a 4-channel memory system. And I have used one memory controller component in sst elements and a Dramsim2 backend to implement it before. That is to say, the Dramsim2 backend was configured to have 4 channels(i.e. 4 memory controllers in Dramsim2). But I don't think it's a reasonable way to implement a 4-channel memory system beacuse of the additional memory controller in the structure.

So I wonder if I can use 4 memory controller components in sst elements to implement the 4-channel memory system. If so, **can I configure it to implement the memory interleaving across channels** ? Or are there some other ways to implement multi-channel memory system using the memory controller component in sst ? I have noticed that the memory controller component has  parameters about memory interleaving([Line 58 and Line 59](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/memHierarchy/memoryController.h)). But I can't figure out it's used by memory interleaving across channels or across ranks.

hughes-c, 2021-07-14:
@SteveY4ng There are several examples in various elements that show how to use interleaving. If you look at <sst>/shogun/tests/hierarchy_test.py, you can see how to use interleave_size and interleave_step in the memory controller connected to either cramsim or simplemem. This is a fairly complex wiring script but includes use cases for multiple components. There are other examples of interleaving in:
memHierarchy/tests/testBackendHBMDramsim.py
memHierarchy/tests/testKingsley.py
memHierarchy/tests/testDistributedCaches.py

You can use grep to find even more examples that may cover your use case.

hughes-c, 2021-07-16:
@SteveY4ng did you have any other questions or can we close this?

SteveY4ng, 2021-07-17:
I’m sorry for not responding to you timely.
Thank you for your help.
I have no other questions.

closed 2021-07-17 by SteveY4ng