sstsimulator/sst-elements #1693: Vanadis with C++ hello world

calewis, 2021-07-14:
Elements sha is 93ba52d

Program: 
```c++ 
#include <cstdio>
int main(){printf("Hello, World!\n");}
```

Compile command:
```
mipsel-linux-musl-g++ -g prog.cpp -static
```

Trying to run Vanadis with this binary copying the basic_vanadis.py file and swapping the executables appears to hang:
```
```

calewis, 2021-07-14:
@nmhamster I don't have the ability to assign this issue, but I suspect you are the right person.

calewis, 2021-07-14:
If I run for longer it eventually fails:
```
[Core:   0]: Error has been detected in retired instruction. Retired register status:
[Core:   0]: Integer Registers (Count=34)
[Core:   0]: | isa:    0 -> phys:  127 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    1 -> phys:  126 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    2 -> phys:  125 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    3 -> phys:  124 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    4 -> phys:  123 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    5 -> phys:  122 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    6 -> phys:  121 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    7 -> phys:  120 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    8 -> phys:  119 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:    9 -> phys:  118 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   10 -> phys:  117 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   11 -> phys:  116 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   12 -> phys:  115 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   13 -> phys:  114 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   14 -> phys:  113 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   15 -> phys:  112 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   16 -> phys:  111 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   17 -> phys:  110 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   18 -> phys:  109 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   19 -> phys:  108 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   20 -> phys:  107 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   21 -> phys:  106 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   22 -> phys:  105 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   23 -> phys:  104 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   24 -> phys:  103 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   25 -> phys:  102 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   26 -> phys:  101 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   27 -> phys:  100 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   28 -> phys:   99 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   29 -> phys:   98 | r:    0 | w:    0 | v: 0x000000007ffffe40 | v: 2147483200 / 2147483200
[Core:   0]: | isa:   30 -> phys:   97 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   31 -> phys:   96 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   32 -> phys:   95 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:   0]: | isa:   33 -> phys:   94 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
FATAL: [Core:   0]: Instruction 0xa0100000 flags an error (instruction-type=DECODE_FAULT) at cycle 8750912
```

nmhamster, 2021-07-14:
@calewis - can you try the C equivalent (not C++) please?

calewis, 2021-07-14:
Already did, same issue

nmhamster, 2021-07-14:
What machine are you using?

calewis, 2021-07-14:
RHEL 7 Desktop.

nmhamster, 2021-07-14:
Can you get me the version of your MUSL toolchain?

nmhamster, 2021-07-14:
BTW - the execution never starts (all registers cleared implies the entry point it broken).

calewis, 2021-07-14:
It's musl-cross-make with config.make.dist having
```
TARGET=mipsel-linux-musl
```

calewis, 2021-07-14:
The issue ended up being a configuration issue on my end.  I had accidentally built a mips compiler instead of mipsel compiler.

nmhamster, 2021-07-14:
Querying the type of the file should give the following (note - LSB is needed for Little-Endian):

```$ file sqrt-musl```

Gives:

```sqrt-musl: ELF 32-bit LSB executable, MIPS, MIPS-I version 1 (SYSV), statically linked, with unknown capability 0xf41 = 0x756e6700, with unknown capability 0x70100 = 0x1040000, not stripped```

nmhamster, 2021-09-13:
Closed - endian-ness of binary is now detected during ELF loading.

closed 2021-09-13 by nmhamster