sstsimulator/sst-elements #1694: C++ Hello World Iostream Hangs

calewis, 2021-07-14:
Trying to run:
```c++
#include <iostream>
int main(){std::cout << "Hello, World!\n" << std::flush;}
```

Runs for a very very long time.  Here is cycle 98700 (although I have run about a million cycles without finishing):
```
[Core:   0]: ============================ Cycle        98700 ============================
[Core:   0]: -- Core Status:
[Core:   0]: ---> Thr:     0 (unhalted) / ROB-Pend: 9 / IntReg-Free: 89 / FPReg-Free: 133
[Core:   0]: -- Resetting Zero Registers
[Core:   0]: => Fetch Stage <==========================================================
[Core:   0]: => Decode Stage <==========================================================
[Core:   0]: -> Decode step for thr: 0
[Core:   0]: ---> Max decodes per cycle: 2
[Core:   0]: Instruction Loader - Internal State Report:
[Core:   0]: --> Cache Line Width:          64
[Core:   0]: --> Pending Loads:             0
[Core:   0]: --> uop Cache Entries:         1536 / 1536
[Core:   0]: --> Predecode Cache Entries:   4 / 4
[Core:   0]: ---> Found uop bundle for ip=0x04e1b78, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 9 entries.
[Core:   0]: ---> Instruction for issue is not a branch, continuing with normal copy to issue-queue...
[Core:   0]: ---> --> issuing ins addr: 0x04e1b78, ADDI...
[Core:   0]: ---> Found uop bundle for ip=0x04e1b7c, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 10 entries.
[Core:   0]: ---> Instruction for issue is not a branch, continuing with normal copy to issue-queue...
[Core:   0]: ---> --> issuing ins addr: 0x04e1b7c, OR...
[Core:   0]: ---> Performed 0 decodes this cycle, 2 uop-bundles used / updated-ip: 0x4e1b80.
[Core:   0]: -> Decode step for thr: 0
[Core:   0]: ---> Max decodes per cycle: 2
[Core:   0]: Instruction Loader - Internal State Report:
[Core:   0]: --> Cache Line Width:          64
[Core:   0]: --> Pending Loads:             0
[Core:   0]: --> uop Cache Entries:         1536 / 1536
[Core:   0]: --> Predecode Cache Entries:   4 / 4
[Core:   0]: ---> Found uop bundle for ip=0x04e1b80, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 11 entries.
[Core:   0]: ---> Instruction for issue is not a branch, continuing with normal copy to issue-queue...
[Core:   0]: ---> --> issuing ins addr: 0x04e1b80, OR...
[Core:   0]: ---> Found uop bundle for ip=0x04e1b84, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 12 entries.
[Core:   0]: -----> Last instruction in the bundle causes potential branch, checking on branch delay slot
[Core:   0]: ---> Proceeding with issue the branch and its delay slot...
[Core:   0]: ---> --> issuing ins addr: 0x04e1b84, BCMPIL...
[Core:   0]: ---> Branch 0x4e1b84 predicted taken, jump to 0x4e179c
[Core:   0]: ---> Forcing IP update according to branch prediction table, new-ip: 4e179c
[Core:   0]: ---> --> issuing ins addr: 0x04e1b88, OR...
[Core:   0]: ---> Performed 0 decodes this cycle, 3 uop-bundles used / updated-ip: 0x4e179c.
[Core:   0]: -> Decode step for thr: 0
[Core:   0]: ---> Max decodes per cycle: 2
[Core:   0]: Instruction Loader - Internal State Report:
[Core:   0]: --> Cache Line Width:          64
[Core:   0]: --> Pending Loads:             0
[Core:   0]: --> uop Cache Entries:         1536 / 1536
[Core:   0]: --> Predecode Cache Entries:   4 / 4
[Core:   0]: ---> Found uop bundle for ip=0x04e179c, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 14 entries.
[Core:   0]: ---> Instruction for issue is not a branch, continuing with normal copy to issue-queue...
[Core:   0]: ---> --> issuing ins addr: 0x04e179c, SYNC...
[Core:   0]: ---> Found uop bundle for ip=0x04e17a0, loading from cache...
[Core:   0]: -----> Bundle contains 1 entries.
[Core:   0]: ----> thr-rob contains 15 entries.
[Core:   0]: ---> Instruction for issue is not a branch, continuing with normal copy to issue-queue...
[... 484 lines cut ...]
[Core:   0]: ----> ROB[ 3]: ins: 0x00000000004e1b60 /       ANDI / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 2]: ins: 0x00000000004e1b5c /       LOAD / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 1]: ins: 0x00000000004e1b58 /        NOR / error:  no / issued: yes / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 0]: ins: 0x00000000004e1b54 /       LOAD / error:  no / issued: yes / spec:  no / rob-front: yes / exe:  no
[Core:   0]: -- ROB: 18 out of 64 entries:
[Core:   0]: ----> ROB[17]: ins: 0x00000000004e17a8 /        ADD / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[16]: ins: 0x00000000004e17a4 /        NOP / error:  no / issued: yes / spec:  no / rob-front:  no / exe: yes
[Core:   0]: ----> ROB[15]: ins: 0x00000000004e17a0 /  LLSC_LOAD / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[14]: ins: 0x00000000004e179c /       SYNC / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[13]: ins: 0x00000000004e1b88 /         OR / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[12]: ins: 0x00000000004e1b84 /     BCMPIL / error:  no / issued: yes / spec: yes / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[11]: ins: 0x00000000004e1b80 /         OR / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[10]: ins: 0x00000000004e1b7c /         OR / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 9]: ins: 0x00000000004e1b78 /       ADDI / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 8]: ins: 0x00000000004e1b74 /      STORE / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 7]: ins: 0x00000000004e1b70 /     BCMPIL / error:  no / issued: yes / spec: yes / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 6]: ins: 0x00000000004e1b6c /         OR / error:  no / issued: yes / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 5]: ins: 0x00000000004e1b68 /         OR / error:  no / issued: yes / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 4]: ins: 0x00000000004e1b64 /         OR / error:  no / issued: yes / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 3]: ins: 0x00000000004e1b60 /       ANDI / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 2]: ins: 0x00000000004e1b5c /       LOAD / error:  no / issued:  no / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 1]: ins: 0x00000000004e1b58 /        NOR / error:  no / issued: yes / spec:  no / rob-front:  no / exe:  no
[Core:   0]: ----> ROB[ 0]: ins: 0x00000000004e1b54 /       LOAD / error:  no / issued: yes / spec:  no / rob-front: yes / exe:  no
[Core:   0]: ================================ End of Cycle ==============================
```

calewis, 2021-07-14:
Without the verbosity so high I can easily get to multiple ms of simulation time with no result.

nmhamster, 2021-07-14:
Can you set the verbose parameter on the **os** component and copy in the output (not the pipeline verbose)?

calewis, 2021-07-14:
```
[Core:   0]: ============================ Cycle       183905 ============================
[os]: System Call (syscall-ins: 0x4e1684, link-reg: 0x4eb4f0)
[os]: --> [SYSCALL-handler] syscall-ins: 0x4e1684 / call-code: 4238
[os]: [syscall-handler] found a call to futex( 0x52fc78, 128, 2147483664, 0, sp: 0x7ffffc38 (arg-count is greater than 4))
[os]: -> recv os response
[os]: syscall return-code: 0 (success: yes)
[os]: -> issuing call-backs to clear syscall ROB stops...
[os]: syscall - generating successful (v: 0) result (isa-reg: 7, phys-reg: 90)
[Core:   0]: ================================ End of Cycle ==============================
```

nmhamster, 2021-11-17:
Working in most up to date Vanadis code. Reopen if encountered again.

closed 2021-11-17 by nmhamster