sstsimulator/sst-elements #1712: c_AddressHasher.hpp missing include

calewis, 2021-08-02:
New Issue for sst-elements
--------------------------

1 - CramSim.cpp fails to build due to undefined uint type
```
 error: 'uint' was not declared in this scope; did you mean 'int'?
            void parsePattern(std::string *x_inStr, std::pair<std::string, uint> *x_outPair);
  ```

3 - alpine linux docker image

The header `sys/types.h` is available but not included.  This was fixed by just including that header.

calewis, 2021-08-02:
Although `uint` isn't considered portable and should really be changed to unsigned int. Or the typedef should be provided by the library itself.

calewis, 2021-08-02:
c_TxnGen.cpp has the same issue

calewis, 2021-08-02:
As does GNA.h

calewis, 2021-08-02:
ariel_shmem.h is missing sys/time.h

hughes-c, 2021-08-02:
I remember fixing some of those CramSim types a year or so ago but I guess I never pushed them.

closed 2023-10-25 by hughes-c