sstsimulator/sst-elements #1720: Packet Injection Delay in AlltoAll Motifs

hanskasan, 2021-09-03:
Hi,

I was trying to run alltoall motif and noticed that there are delays between packet injections. I printed out the time when LinkControl at node 0 send the events to PortControl ('LinkControl::send()') and obtained the following timeline.

```
EMBER: using param directory: paramFiles
EMBER: platform: default
EMBER: network: topology=dragonfly shape=2:4:1:9
EMBER: numNodes=72 numNics=72
EMBER: network: BW=4GB/s pktSize=2048B flitSize=8B
EMBER: Job=0, nidList='0-71'
EMBER: Motif='Init'
EMBER: Motif='Alltoall iterations=1 compute=0 bytes=8'
EMBER: Motif='Fini'
Send on LinkControl in NIC from: 0 to: 1 at: 554000
Send on LinkControl in NIC from: 0 to: 2 at: 2016000
Send on LinkControl in NIC from: 0 to: 3 at: 3190000
Send on LinkControl in NIC from: 0 to: 4 at: 4220000
Send on LinkControl in NIC from: 0 to: 5 at: 5552000
```

There are at least 10,000 cycles of delay between the events. I don't think these delays should be necessary in an alltoall application, i.e. node 0 can proceed to send packets to node 2 right after sending to node 1.

May I know what causes these delays to happen? Is it possible to model the alltoall application without these delays?

I ran the simulation with the following configuration:
run `sst emberLoad.py`
Motif at defaultSim.py is configured as `motif['cmd'] = "Alltoall iterations=1 compute=0 bytes=8"`

I ran the SST on both Ubuntu 18.04.5 LTS and CentOS 7.5.1804.

I built the SST from the distributed SST Core 11.0.0 and SST Elements 11.0.0 tarfiles (2021-May-03 release). 

Thank you!

feldergast, 2021-09-07:
I don't see those prints in the merlin LinkControl or FireflyNic classes.  Have you added them yourself?  Given that the times are all multiples of 1000, they look like they are in picoseconds (the default timebase of the simulation).  What call did you use to get the simulated time to print?  If you used getCurrentSimCycle(), then it is definitely in picoseconds.  You can use getCurrentSimTimeNano() to get the time in nanoseconds.

If I divide those times by 1000 and compute the time between, you end up with times that are on the order of twice the serialization time of a 2048B packet at 4GB/s bandwidth.  I'd have to know where the print statement is in order to drill down any deeper.

hanskasan, 2021-09-08:
I apologize for the confusion.

Yes, I added the prints by myself. The printout code is in the last lines of the code, right above `return true`.

`bool LinkControl::send(SimpleNetwork::Request* req, int vn) {

    // Check to see if the VN is in range
    if ( vn >= req_vns ) return false;
    req->vn = vn;

    // Check to see if we need to do a nid translation
    if ( use_nid_map ) req->dest = nid_map[req->dest];

    // Get the output queue information for that vn
    output_queue_bundle_t& out_handle = *(vn_remap_out[vn]);

    // Do the VN remapping
    int real_vn = out_handle.vn;

    // Create a router event using id and original vn
    RtrEvent* ev = new RtrEvent(req,id,real_vn);
    // Fill in the number of flits
    ev->computeSizeInFlits(flit_size);
    int flits = ev->getSizeInFlits();

    // Check to see if there are enough credits to send
    if ( out_handle.credits < flits ) return false;

    // Update the credits
    out_handle.credits -= flits;
    // ev->request->vn = vn;

    ev->setInjectionTime(getCurrentSimTimeNano());
    out_handle.queue.push(ev);
    if ( waiting && !have_packets ) {
        output_timing->send(1,nullptr);
        waiting = false;
    }

    // If the dest is throttled, record the packet
    if ( congestion_state.count(req->dest) != 0 ) congestion_state[req->dest].count++;

    if ( ev->getTraceType() != SimpleNetwork::Request::NONE ) {
        output.output("TRACE(%d): %" PRIu64 " ns: Send on LinkControl in NIC: %s\n",ev->getTraceID(),
                      getCurrentSimTimeNano(), getName().c_str());
    }

    if (req->src == 0){
        std::cout << "Send on LinkControl in NIC from: " << req->src << " to: " << req->dest << " at: " << getCurrentSimCycle() << std::endl;
    }

    return true;
}`

The times are in picoseconds.

Thank you.

hanskasan, 2021-09-17:
Hi,

I apologize but if possible, can I get an update on this?

It seems that the delay might be caused by the alltoall library in SST being implemented using "blocking" behavior. However, I am not sure if this is necessary as there should be no reason why alltoall should require "blocking".

Thank you.

feldergast, 2021-09-21:
I didn't write any of the ember/firefly code, so I'm not clear on all the details of the alltoall implementation.  However, looking at the code, the alltoall is not a point-to-point implementation where everyone sends to everyone, but is grouping ranks and exchanging data in phases.  The implementation accounts for the overheads of copying the data into buffers and posting both sends and receives.  It looks like it posts a receive, sends its data, then waits for the other data to arrive.  It then does this again with a larger data chunk.  It has to wait on the receive because it needs that data so it can send it in the next phase.  I think the delays you're seeing are the waits between phases.  I'll see if I can get the implementor of that code to confirm.

mjleven, 2021-09-21:
The all to all algorithm is a simple ring. Node N receives from N-1 and sends to N+1 for round 1. Round 2 is N-2 and N+2.

hanskasan, 2021-09-23:
Thank you very much for the reply!

May I know which "alltoall" code are you referring to?

I tried to look at `emberalltoallev.h` but it does not have any information on how the alltoall is implemented.

Thanks.

mjleven, 2021-09-23:
I was looking at firefly/funcSM/alltoallv.cc. 

emberalltoallev.h describes the event that gets sent to firefly.

hanskasan, 2021-10-01:
I see. Thank you very much for your help!

closed 2021-10-01 by hanskasan