sstsimulator/sst-elements #1721: New component is not recognized by SST

francisco-munoz, 2021-09-13:
Hello,

I am trying to integrate a new component to SST, but the component seems  not to be recognized. 

The new component is called STONNE and is a new simulator. I have created a new folder called sstStonne in myLocation/sst-elements/src/sst/elements. Inside it, I have placed the simulator itself along with two files that are supposed to wrap the simulator. These two files are called sstStonne.cc and sstStonne.h. In sstStonne.h I have done the necessary to create the wrapper:

```
namespace SST {
namespace SST_STONNE {
class sstStonne : public SST::Component {
public:
     // REGISTER THIS COMPONENT INTO THE ELEMENT LIBRARY
    SST_ELI_REGISTER_COMPONENT(
        sstStonne,
        "sstStonne",
        "MAERI",
        SST_ELI_ELEMENT_VERSION(1,0,0),
        "Configurable Dataflow Component",
        COMPONENT_CATEGORY_PROCESSOR
    )

    SST_ELI_DOCUMENT_PARAMS(
       // { "fp_mul_latency", "Number of clock ticks for FP MUL operations", "4" },
       // { "fp_div_latency", "Number of clock ticks for FP DIV operations", "4" }
    )

    ///TODO
    SST_ELI_DOCUMENT_STATISTICS(
    )

    SST_ELI_DOCUMENT_PORTS(
       // { "cache_link",     "Link to Memory Controller", { "memHierarchy.memEvent" , "" } }
    )

    SST_ELI_DOCUMENT_SUBCOMPONENT_SLOTS(
        //{ "memory",         "The memory interface to use (e.g., interface to caches)", "Experimental::Interfaces::SST::StandardMem" }
    )
    sstStonne(SST::ComponentId_t id, SST::Params& params);
    sstStonne();
    ~sstStonne();
   ....

```
The .cc code implements the functions tic(), setup() and finish().

Then, I created the Makefile based on simpleElementExample:

nmhamster, 2021-09-13:
@francisco-munoz - are you able to see `sstStone` listed when you run `sst-info` to dump the component listing that the SST sub-system sees?

What errors are output by SST when you attempt to run `sst` with a model?

francisco-munoz, 2021-09-13:
Hi @nmhamster , 

thanks for the quick reply. I posted it without haven't finished it by mistake. Sorry about that. I have updated the post. As you see, I have run sst-info, but the system says it cannot find the component.

nmhamster, 2021-09-13:
@francisco-munoz - can you run:

`ls /home/paco/Desktop/SST-Project/sst-elements/sst-elements-install/lib/sst-elements-library` 

For us and see what files you have in there?

Also, do any of the basic models and tests run correctly right now (this would confirm that some elements are installed and working in general).

francisco-munoz, 2021-09-13:
Hi again @nmhamster , thanks.

This is the result of the ls:

_[paco@paco-pc tests]$ ls /home/paco/Desktop/SST-Project/sst-elements/sst-elements-install/lib/sst-elements-library
libsimpleElementExample.la  libsimpleElementExample.so  libsstStonne.la  libsstStonne.so
[paco@paco-pc tests]$_ 

Both elements I have installed (sstStonne and simpleElementExample) seem to appear on the list, but only simpleElementExample works. So yes, that basic component is modeled and working in the system, so I think it's not something general. Here is the proof I am simulating simpleElementExample:

_[paco@paco-pc tests]$ sst example0.py 
Simulation is complete, simulated time: 52 ns_

Also, simpleElementExample is in the list when running sst-info -l. 

To further clarify, I had all the components installed previously and were passing the tests. I just removed them to remove the noise regarding the rest of the components.

Thank you

francisco-munoz, 2021-09-13:
In case it helps, I have pushed the changes on my fork, in branch sst-stonne:

gvoskuilen, 2021-09-13:
@francisco-munoz I sometimes see this behavior when I'm missing a function implementation. You have two functions declared in sstStonne.h: sstStonne() and ~sstStonne() but I don't see the implementation of these in the class. Does defining these fix the issue? 

```
sstStonne() { }
~sstStonne() { }
```

francisco-munoz, 2021-09-13:
@gvoskuilen Wow.. that was the problem! I was missing the implementation of the destructor (i.e., ~sstStonne()).. after implementing a (empty) destructor SST recognizes the new component. Thank you very much for the support. Really appreciated!!.

nmhamster, 2021-09-13:
Wahoo thanks @gvoskuilen!

closed 2021-09-13 by nmhamster