sstsimulator/sst-elements #1754: Vanadis is too fast

calewis, 2021-11-18:
@nmhamster Vanadis was running too fast so I needed to slow it down a bit.  I added the following to my empty main
```c
#include <unistd.h>
int main(){
  sleep(1);
}
```
to make it take longer, but
```
FATAL: [os]: [syscall-handler] Error: unknown code 4166 (ins: 0x401584, link-reg: 0x4014d4)
```
prevents my simulation from finishing.

nmhamster, 2021-11-18:
@calewis - you are the first user to complain the code is too fast :-). I'll take a look, this is just an OS call we need to add I believe.

nmhamster, 2021-12-01:
Do you really want to sleep for long periods (like a second or more?) or are you looking for smaller sleep periods that are functionally correct with respect to the sleep time? Asking the question because there is the potential to optimize differently depending on the answer.

calewis, 2021-12-01:
I was actually just testing functionality.  

```c++
auto t0 = std::chono::system_clock::now();
sleep(1);
auto time = std::chono::system_clock::now() - t0;
```
should lead to a duration that is approximately the amount of time the sleep was suppose to take.

hughes-c, 2023-06-22:
@nmhamster @mjleven Is this still a thing?

hughes-c, 2023-10-26:
This still fails with
`FATAL: [os_hdlr]:handleSysCall() Error: unknown code 115 (ins: 0x38554, link-reg: 0x64)`