sstsimulator/sst-elements #1765: Event queue empty while simulating Ember MPI Sweep3D

saichenna, 2021-11-29:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
     I'm currently using SST to perform network simulations. While testing various available Ember MPI communication patterns on different topologies, some Ember patterns are simulating infinite time stating "*** Event queue empty, exiting simulation... ***
Simulation is complete, simulated time: 18.4467 Ms". Note that the simulation ran successfully for Collective benchmarks such as Allreduce, Broadcast, Scatter. However patterns like Sweep3D are specifying infinite simulation time stating that the event queue is empty. I'm attaching the configuration file below. Is there supposed to be a different way to add motifs in the configuration file?

2 - Describe how to reproduce
I have tweaked the dragon_128_allreduce.py configuration file specified in the Ember tests folder to include Sweep3D motif. I have attached the configuration file. 
3 - What Operating system(s) and versions 
Ubuntu 20.04.3 LTS
4 - What version of external libraries (Boost, MPI)
OpenMPI 4.0.6
5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)

6 - Fill out Labels, Milestones, and Assignee fields as best possible
[dragon_128_allreduce.txt](https://github.com/sstsimulator/sst-elements/files/7621424/dragon_128_allreduce.txt)
[dragon_128_allreduce.txt](https://github.com/sstsimulator/sst-elements/files/7621427/dragon_128_allreduce.txt)

saichennaintel, 2021-12-07:
Any help would be greatly appreciated.

feldergast, 2021-12-07:
The problem is your buffer sizes, which are set to 1kB.  It turns out that the firefly stack has an MTU of 1024 bytes, so the packets are too large to send.  Thus no events get sent, so the simulation ends.  I would suggest changing the buffer sizes to about 14kB.

saichennaintel, 2021-12-08:
Thank you @feldergast! It works fine now.