sstsimulator/sst-elements #1787: Question: How to enable avx512 support in ariel-cpu model?

luckyq, 2022-01-10:
I have read the introduction of Ariel CPU in sst-simulator.org. It said:

"Users can also build their application with any specialized optimization or instruction set flags (e.g. SSE, AVX, AVX-2) as required, the front ends will correctly interpret the memory requests and encode these for the emulation core.".

Does this mean we can use some intrinsics such as SIMD functions in Ariel CPU simulator, the SIMD (for example, add -fsmdavx512 as one flag in G++ command)?

And if it supports it, are there some changes that should be made?

Thanks in advance.

Best,
Dong

hughes-c, 2022-02-03:
I'm fairly certain (not 100%) that the 2.14 tool does not have support for AVX-512 instructions because the runtime that ships with it is so old. However, the 3.x tool does.

luckyq, 2022-02-04:
Thanks for the reply. I have one more question.
Does Ariel really simulate the instructions or just extract the memory access for the instructions? 

Thanks,
Dong

hughes-c, 2022-02-04:
Ariel is not a pipeline model -- everything runs natively on the host machine (via the JIT) and the memory accesses are offloaded to the detailed memory models. We have a pipeline model, vanadis, that runs MIPS32 and RV64.

nmhamster, 2022-02-07:
PIN 2.14 supports the AVX512 instruction set which worked on KNL, not necessarily the same instruction sets which work on other Intel products. In general these are similar but it might lead to some unknown instructions being encountered.