sstsimulator/sst-elements #1813: Inexplicable high latency when using adaptive routing with Fat Tree

tommasobo, 2022-01-28:
I have been struggling for a couple of days with this weird issue where the adaptive routing mechanism in Fat Tree results in packets having a much higher latency on average compared to the deterministic routing. In particular few of them arrive with an insane 20-30x higher latency when compared to deterministic routing. 

I am running SST 11 on Ubuntu 20.04 but the issue has also been recreated in other OS and with a fresh install of SST. To go more into details of what I do:

1) I have a motif where each node (topology is 32:32:32 dragonfly but also tried others like 32:16:32) does a bunch of irecv and isend to other randomly selected nodes. I have a csv file where i generate beforehand the target nodes for the various irecv and isend. The messages are of random size but usually below 1 MiB. Between isend, there is a compute of x ns based on the message size to change the offered load. All the packets arrive correctly at the right destination.

2) After that I add some simple variables (I think to internal_router_event and FireflyNetworkEvent) to track the ingress time and egress time. To make double sure my timings are correct I calculate two independent latencies (one NIC->NIC and the other Ingress Switch -> Egress Switch for each packet).

3) After that I simply analyze the results and each time I notice the deterministic one is better for both latency and throughput . Moreover I can see that while the deterministic packet have a somewhat small distribution, for the adaptive one some packets arrive incredibly late as previously said. The weird this is that those packets pick ports which in theory still have a lot of credits left (that is the point of adaptive after all) so I am not sure how this is happening.

It's possible the problem is also a simple mistake on my side but I have tried to use a fresh installation of SST to reduce possible errors and I have it checked by other people too. If anyone knows if this is the expected behavior or has any clue about what is happening, then it would be a huge help! Thanks

feldergast, 2022-01-28:
Are the inconsistencies you are seeing in Fat tree or dragonfly (both are mentioned).  If fat tree, the adaptive routing implementation is experimental and has not been verified/hardened as fat tree has not been a focus of our research.  We generally used the deterministic routing on a fat tree as a stable baseline to compare other topologies. 

If the problem is in dragonfly, then the adaptive routing algorithms have been tested on a variety of test cases and have seemed to work appropriately.

Other notes:  if you're sending large messages (1 MiB qualifies), the network could be congesting depending on how often data is being sent.  A congested network can easily add 20x latency to a packet.

tommasobo, 2022-01-28:
So the issue seems to be mostly with Fat Trees (I will verify later once again that everything works correctly on Dragonfly).

I have a very simple setup to explain this Fat Tree problem. I have a 32,32:32 topology, I ask the first 32 nodes to send data to the last 32 nodes (0 to 1023 , 1 to 2022....).  I then remove the "dest" parameter when doing deterministic routing for the up routes (basically to force the first 32 nodes to take the same up link). As expected, I get an average flow bandwidth which is 1/32 of my global bandwidth. So far it makes sense. The problem is that when I try adaptive with the same setup, it does the exact same thing when it should clearly pick other ports. It seems that for some reason the single port which is being utilized always have enough credits (as in, its credits get increased before the next "packet" is considered and hence it thinks it is the best one to pick). To fix these problems I noticed two things:

1. Obviously the easy fix is to just put back the dest parameter when taking up routers to take a different port for each flow. But this isn't really what I want and removing dest was just to show the issue in the first place.
2. I noticed that setting the xbar_bw to a number much greater than the link_bw allows adaptive routing to work again. It's as if with xbar_bw  = link_bw that each packet is treated sequentially and hence the credits for that port are always put back before the next one is considered.

Thanks!

tommasobo, 2022-02-02:
Just to give some more information. From our testing it seems the credits don't take into consideration whether there are many packets going through the same port or not. Let's have again a simple example:

We have this simple setup where we have three hosts connected to a leaf switch sending data to three destinations connected to another leaf switch.

1) At the first time step, the sending switch receives 1 packet from each of the NICs. Let's say we have a dumb routing algorithm which always select port 5 (which is a up port) for these 3 packets.  This should create a small congestion on that port 5 going up.

2) The NICs are still sending packets, so the switch in its second "reading cycle" receives three more packets like before. At this point I print the credits and the output_queue_length for port 5 and, to my surprise, the credits are not decreasing (same thing for the output_queue_length which is not increasing). 

So I don't understand why these metrics don't detect that there are many packets trying to send data to the same port over time. What do these credits even detect then? Just if a port is busy at a given time? Is there a way to track what I want here?

Thanks

feldergast, 2022-02-02:
I suspect that what's happening is that the xbar_bw and the link_bw are set at the same value.  This means that the output queue length will remain constant as long as the input buffers are targeting that port.  Essentially, one packet is sent on the output link and one packet is put into the output queue.  The credits across the xbar would also stay constant for the same reason.  When I configure simulations, I usually make the xbar_bw approximately 1.5x the link_bw.  A typical switch will have some speedup factor over it's crossbar compared to the incoming bandwidth on the ports.

closed 2022-03-09 by tommasobo