sstsimulator/sst-elements #1814: Understanding MemH statistics

francisco-munoz, 2022-01-31:
Hi, 
I am not able to understand completely the statistics being reported by SST. Could anyone help me out? I am trying to get the Cache Misses from the Accumulator Statistics generated by MemHierarchy. It seems, for example, the stats associated with the misses for the getS() event are like this:
l1cache.latency_GetS_miss : Accumulator : SimTime = 910475000; Rank = 0; Sum.u64 = 15379174; SumSQ.u64 = 483900274; Count.u64 = 557083; Min.u64 = 11; Max.u64 = 67;
The question is basically, what is actually the number of misses? Sum.u64 or Count.64? Also what does Min.u64 and Max.u64 mean?
Also, do you know how can I get the total number of requests sent to the cache?

Best regards.

hughes-c, 2022-02-03:
@francisco-munoz Count is the number of events; Sum is the total latency for GetS misses. The min/max values are the min and maximum latency reported for this statistic.

For total requests, I'm not sure. @gvoskuilen is it TotalEventsReceived?

francisco-munoz, 2022-02-14:
Thank you @hughes-c. It works!

closed 2022-02-14 by hughes-c