sstsimulator/sst-elements #1825: SST for Trace Replay

hanskasan, 2022-02-11:
Hi,

I am planning to do trace replay evaluations using SST. Based on the tutorials provided on the website, seems like this can be done using both SST-Core (with Zodiac) and SSTMacro. However, since I need a cycle-accurate network simulator, which I think is only available in SST-Core (Merlin), the only available option is to use Zodiac but I cannot find any documentation on it. Will it be possible for me to get some basic documentation on how to use the Zodiac element?

Thank you!

jpkenny, 2022-02-14:
You are correct that sst-macro does not support cycle-accurate network simulation.

Is there appropriate sst-elements documentation that somebody can point to?

nmhamster, 2022-02-24:
@hanskasan - is this for MPI applications?

hanskasan, 2022-02-25:
Yes, I am planning to DUMPI trace replay in particular.

nmhamster, 2022-02-25:
@hanskasan - do you have to use DUMPI or will other profiling/tracing tools also work? Zodiac DUMPI support is not fully complete, hence asking if other tools may be more applicable.

hanskasan, 2022-08-31:
Hi, sorry but can I get an update on this?
I need to run DUMPI trace on a cycle accurate network simulator e.g. Merlin. Is there a workaround for this? If I link both SST-Macro and Merlin to SST-Core, will I be able to run the DUMPI trace on Merlin? Thank you.

sumant-kalra, 2024-03-19:
@nmhamster @gvoskuilen @allevin   What other tracing tools are supported by Zodiac? I see that you have a sirius trace reader in zodiac that I used to run the test case with your sample trace file downloaded from sst_downloads.