sstsimulator/sst-elements #1862: Configure Errors after pulling recent changes

plavin, 2022-04-05:
Issue: 

I'm now getting the following configure errors when trying to build sst-elements. I've got the devel branches of both sst-core and sst-element running. 

When I try to configure sst-elements, I get the following error:

```
./configure: line 15990: syntax error near unexpected token `config_memHierarchyExt=1,config_memHierarchyExt=0'
./configure: line 15990: `   SST_memHierarchyExt_CONFIG(config_memHierarchyExt=1,config_memHierarchyExt=0)'
```

I'm running on Ubuntu 20.04, fully updated.

sst-core: 6648eed0a27b1654c13f5a193a203a383a2503b4
sst-elements: 1234a40abfcbb82046f24d47cdc7a2a8f3fa29b2

Any thoughts on what is causing this?

nmhamster, 2022-04-05:
Have you re-run `./autogen.sh` since updating?

plavin, 2022-04-05:
Yes, I've done that. Is there some other state that may be breaking things? Are there some files I should try deleting?

nmhamster, 2022-04-05:
Is this an entirely fresh directory (where you are configuring)?

plavin, 2022-04-05:
I had tried cleaning things up some but I guess I didn't do enough. Pulling the copy from the official repo worked. I guess I've got something broken with my forked copy. Thanks for the help.