sstsimulator/sst-elements #1893: Simulating Multi-threaded Program on Vanadis

Connie120, 2022-04-26:
Hello! I'm trying to add this functionality to run multiple threads in parallel on Vanadis, and given the current situation that Vanadis is not able to simulate pthread programs due to some unimplemented syscalls, I'm thinking loading a single-thread program and expand it into multiple threads to test for now. Any advice on how should I approach this in a better way? Any help would be appreciated, thank you!

plafratt, 2023-10-17:
Is this resolved? In case it isn't, I thought I should make a note that I was able to successfully use pthread_create(). There is a test in the repository in `vanadis/tests/small/misc/pthread/` that I found helpful.

Connie120, 2023-10-18:
@plafratt This has been resolved, thanks for mentioning!

closed 2023-10-18 by Connie120