sstsimulator/sst-elements #1906: Ember: Failed Simulations for EmberCMT3D & EmberCMTCR. Event Queue empty

saichenna, 2022-05-16:
I'm trying to run Ember CMT Benchmarks on different network topologies. However, while running the network simulation, I encounter the following error: 
* Event queue empty, exiting simulation... *

I'm attaching the configuration file below. I'm also including the simulator output.
[image]
[dragon_32_cmt3d.txt](https://github.com/sstsimulator/sst-elements/files/8703609/dragon_32_cmt3d.txt)

feldergast, 2022-05-17:
The error you're seeing happens when the event queue has no more events to process.  Usually this means that no initial events were sent.  I'm not familiar with the CMT motifs, so I'm not sure how they work.  Do things run correctly if you just run init, all reduce, fini?  If so, then the issue is likely in the CMT motif.  If not, we'll have to look a little deeper at the configuration file, but at first glance, it looks fine to me.

saichenna, 2022-05-17:
Yes, I was able to run just the init, allreduce, fini motif with the same configuration file (commenting line 64 and uncommenting line 65 in the attached config file). Looks like the issue is with CMT motif.

saichenna, 2022-05-31:
Hello @feldergast ! Any update on this error. I'm familiar with CMT-nek communication patterns, but i'm not sure how to proceed debugging with this error. Any pointers on where to start looking?