sstsimulator/sst-elements #1909: make error on mac

prateek22sri, 2022-05-27:
[mac:~/Software/sst/sst-elements-library-12.0.0] $ make
Making all in src
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in sst
Making all in elements
Making all in ariel
  CXX      libariel_la-arielcpu.lo
  CXX      libariel_la-arielcore.lo
  CXX      libariel_la-arielmemmgr_simple.lo
  CXX      libariel_la-arielmemmgr_malloc.lo
  CXX      libariel_la-arielevent.lo
  CXX      libariel_la-arieltexttracegen.lo
  CXX      libariel_la-arielgzbintracegen.lo
  CXXLD    libariel.la
Making all in cacheTracer
  CXX      cacheTracer.lo
  CXXLD    libcacheTracer.la
Making all in cassini
  CXX      strideprefetch.lo
  CXX      palaprefetch.lo
  CXX      nbprefetch.lo
  CXX      pageentry.lo
  CXX      addrHistogrammer.lo
  CXX      cacheLineTrack.lo
In file included from cacheLineTrack.cc:20:
cacheLineTrack.h:87:15: error: 'uint' was not declared in this scope; did you mean 'int'?
   87 |     Statistic<uint>* useHisto;
      |               ^~~~
      |               int
cacheLineTrack.h:87:19: error: template argument 1 is invalid
   87 |     Statistic<uint>* useHisto;
      |                   ^
cacheLineTrack.h:89:15: error: 'uint' was not declared in this scope; did you mean 'int'?
   89 |     Statistic<uint>* evicts;
      |               ^~~~
      |               int
cacheLineTrack.h:89:19: error: template argument 1 is invalid
   89 |     Statistic<uint>* evicts;
      |                   ^
cacheLineTrack.cc: In constructor 'SST::Cassini::cacheLineTrack::cacheLineTrack(SST::ComponentId_t, SST::Params&)':
cacheLineTrack.cc:59:34: error: 'uint' was not declared in this scope; did you mean 'int'?
   59 |     useHisto = registerStatistic<uint>("hist_word_accesses");
      |                                  ^~~~
      |                                  int
cacheLineTrack.cc:59:39: error: no matching function for call to 'SST::Cassini::cacheLineTrack::registerStatistic<<expression error> >(const char [19])'
   59 |     useHisto = registerStatistic<uint>("hist_word_accesses");
      |                ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/component.h:15,
                 from cacheLineTrack.h:24,
                 from cacheLineTrack.cc:20:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(SST::Params&, const string&, const string&, bool)'
  405 |     Statistics::Statistic<T>* registerStatistic(
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note:   template argument deduction/substitution failed:
cacheLineTrack.cc:59:39: error: template argument 1 is invalid
   59 |     useHisto = registerStatistic<uint>("hist_word_accesses");
      |                ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/component.h:15,
                 from cacheLineTrack.h:24,
                 from cacheLineTrack.cc:20:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(const string&, const string&)'
  449 |     Statistics::Statistic<T>* registerStatistic(const std::string& statName, const std::string& statSubId = "")
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note:   template argument deduction/substitution failed:
cacheLineTrack.cc:59:39: error: template argument 1 is invalid
   59 |     useHisto = registerStatistic<uint>("hist_word_accesses");
      |                ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/component.h:15,
                 from cacheLineTrack.h:24,
                 from cacheLineTrack.cc:20:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(const char*, const char*)'
  471 |     Statistics::Statistic<T>* registerStatistic(const char* statName, const char* statSubId = "")
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note:   template argument deduction/substitution failed:
cacheLineTrack.cc:59:39: error: template argument 1 is invalid
   59 |     useHisto = registerStatistic<uint>("hist_word_accesses");
      |                ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
cacheLineTrack.cc:61:37: error: no matching function for call to 'SST::Cassini::cacheLineTrack::registerStatistic<uint>(const char [7])'
   61 |     evicts = registerStatistic<uint>("evicts");
      |              ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
In file included from /Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/component.h:15,
                 from cacheLineTrack.h:24,
                 from cacheLineTrack.cc:20:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(SST::Params&, const string&, const string&, bool)'
  405 |     Statistics::Statistic<T>* registerStatistic(
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note:   template argument deduction/substitution failed:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(const string&, const string&)'
  449 |     Statistics::Statistic<T>* registerStatistic(const std::string& statName, const std::string& statSubId = "")
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note:   template argument deduction/substitution failed:
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note: candidate: 'template<class T> SST::Statistics::Statistic<T>* SST::BaseComponent::registerStatistic(const char*, const char*)'
  471 |     Statistics::Statistic<T>* registerStatistic(const char* statName, const char* statSubId = "")
      |                               ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note:   template argument deduction/substitution failed:
cacheLineTrack.cc: In member function 'virtual void SST::Cassini::cacheLineTrack::notifyAccess(const SST::MemHierarchy::CacheListenerNotification&)':
cacheLineTrack.cc:110:17: error: 'uint' was not declared in this scope; did you mean 'int'?
  110 |                 uint touched = iter->second.touched.count();
      |                 ^~~~
      |                 int
cacheLineTrack.cc:111:27: error: request for member 'addData' in '*((SST::Cassini::cacheLineTrack*)this)->SST::Cassini::cacheLineTrack::useHisto', which is of non-class type 'int'
  111 |                 useHisto->addData(touched);
      |                           ^~~~~~~
cacheLineTrack.cc:111:35: error: 'touched' was not declared in this scope
  111 |                 useHisto->addData(touched);
      |                                   ^~~~~~~
cacheLineTrack.cc:112:25: error: request for member 'addData' in '*((SST::Cassini::cacheLineTrack*)this)->SST::Cassini::cacheLineTrack::evicts', which is of non-class type 'int'
  112 |                 evicts->addData(1);
      |                         ^~~~~~~
make[5]: *** [cacheLineTrack.lo] Error 1
make[4]: *** [all-recursive] Error 1
make[3]: *** [all-recursive] Error 1
make[2]: *** [all-recursive] Error 1
make[1]: *** [all] Error 2
make: *** [all-recursive] Error 1

jpkenny, 2022-05-31:
Hi,

This looks like a bug. I’m not sure why your build exposes it but our fairly extensive OS X testing doesn’t.

I’ll get a patch into the sst-elements devel branch asap. You should be able to fix it in your local tree by just replacing “uint” with “unsigned int” anywhere it appears in cacheLineTrack.cc.

Can you supply details on your OSX/XCode/compiler versions so we can understand how this got through testing?

Thanks,
Joe

87 | Statistic* useHisto;
| ^~~~
87 | Statistic* useHisto;
| ^
89 | Statistic* evicts;
| ^~~~
89 | Statistic* evicts;
| ^
59 | useHisto = registerStatistic("hist_word_accesses");
| ^~~~
| int
cacheLineTrack.cc:59:39: error: no matching function for call to 'SST::Cassini::cacheLineTrack::registerStatistic< >(const char [19])'
59 | useHisto = registerStatistic("hist_word_accesses");
| ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(SST::Params&, const string&, const string&, bool)'
405 | Statistics::Statistic* registerStatistic(
| ^~~~~~~~~~~~~~~~~
59 | useHisto = registerStatistic("hist_word_accesses");
| ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(const string&, const string&)'
449 | Statistics::Statistic* registerStatistic(const std::string& statName, const std::string& statSubId = "")
| ^~~~~~~~~~~~~~~~~
59 | useHisto = registerStatistic("hist_word_accesses");
| ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(const char*, const char*)'
471 | Statistics::Statistic* registerStatistic(const char* statName, const char* statSubId = "")
| ^~~~~~~~~~~~~~~~~
59 | useHisto = registerStatistic("hist_word_accesses");
| ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
cacheLineTrack.cc:61:37: error: no matching function for call to 'SST::Cassini::cacheLineTrack::registerStatistic(const char [7])'
61 | evicts = registerStatistic("evicts");
| ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:405:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(SST::Params&, const string&, const string&, bool)'
405 | Statistics::Statistic* registerStatistic(
| ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:449:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(const string&, const string&)'
449 | Statistics::Statistic* registerStatistic(const std::string& statName, const std::string& statSubId = "")
| ^~~~~~~~~~~~~~~~~
/Users/pratsriv/Software/sst/sstcore-12.0.0/INSTALL/include/sst/core/baseComponent.h:471:31: note: candidate: 'template SST::Statistics::Statistic* SST::BaseComponent::registerStatistic(const char*, const char*)'
471 | Statistics::Statistic* registerStatistic(const char* statName, const char* statSubId = "")
| ^~~~~~~~~~~~~~~~~

jpkenny, 2022-05-31:
Also appears in cacheLineTrack.h

jpkenny, 2022-05-31:
It looks like this typedef
memHierarchy/Sieve/sieveController.h:    typedef unsigned int uint;
usually gets picked up by cacheLineTrack despite the header not being explicitly included.
Likely a difference in compiler behavior.

I'll remove appearances of "uint" from cassini. This should fix the bug but I don't have enough details to verify/test with the exact environment that the user encountered the bug with.

prateek22sri, 2022-06-01:
Hi Joe,

I don’t have the issue anymore. 
I think when I got this error, I was using g++-11. But later when I used clang, it worked fine. 
OS is Monterey (12.4) with the M1 Chip.
Xcode 13.4 - Build version 13F17a

Also, could you please suggest me a way to learn SST. I am really having a hard time to learn this tool. I want to be able to build custom components and drive those using the python config file. Currently, I am trying to look at the examples but can’t seem to find any documentation for it.

Thanks,
Prateek

jpkenny, 2022-06-01:
Makes sense. I would've guessed g++ since we typically use the Apple clang compilers on OS X and the linux OSs we test tend to have older gcc default compilers. I have a PR in that should fix the issue in devel once it goes through.

I'll respond to the more general question in the other email thread that you started.

prateek22sri, 2022-06-01:
Thanks a ton!

closed 2022-06-21 by prateek22sri