sstsimulator/sst-elements #1913: Unable to run basic_vanadis.py, possible musl toolchain issue

jwmcfar2, 2022-06-01:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement

**When running 'sst tests/basic_vanadis.py' directly from vanadis folder, it sst tests/basic_vanadis.py**

[image]

**The hello-world file was compiled using musl as such: _mipsel-linux-musl-gcc hello-world.c -O3 -static -o hello-world_**
**Which results in a file of type: "hello-world: ELF 32-bit LSB shared object, MIPS, MIPS-I version 1 (SYSV), dynamically linked, not stripped"**

**I ran all the tests for all the elements and Vanadis is the only one Im encountering an issue with, all others passes. This happens on both Master and Devel branch.**

2 - Describe how to reproduce

- **Follow guide to install Core/Elements: http://sst-simulator.org/SSTPages/SSTBuildAndInstall_12dot0dot0_SeriesDetailedBuildInstructions/
(Except pulled from github, rather than use wget)**
- **Follow guide to install MUSL for MIPS32: http://sst-simulator.org/SSTPages/SSTBuildAndInstall_12dot0dot0_SeriesAdditionalExternalComponents/
(Last entry)**
- **Make c program with MUSL compiler: mipsel-linux-musl-gcc hello-world.c -O3 -static -o hello-world**
- **cd into main Vanadis folder, and run sst tests/basic_vanadis.py**

3 - What Operating system(s) and versions 

**Ubuntu 20.04**

4 - What version of external libraries (Boost, MPI)

**mipsel-linux-musl-gcc (GCC) 11.2.1 20211120**

jpkenny, 2022-06-01:
Our tests are running against older versions of the musl toolchains. I get errors using the current toolchains on musl.cc (mipsel and riscv64), but my errors are different from yours.

I produced working toolchains using musl-cross-make (https://github.com/richfelker/musl-cross-make) so that's probably the best bet. It's a long build but it was actually really easy to get musl-cross-make working.

William-An, 2022-06-02:
I also encounter similar issue with the newest MUSL compiler, switching to an older GCC version (10.2.1) help resolve the issue. The link to the older compiler source is [here](https://more.musl.cc/10/x86_64-linux-musl/)

hughes-c, 2022-06-06:
We're actively working on resolving issues with glibc, which should provide support for llvm and gcc compilers.

jpkenny, 2022-06-30:
> Thanks all for your insights - I have made what I feel to be progress, but am still running into issues:
>
> I have compiled musl-cross-make using Target = mipsel-linux-musl, whereas before I simply had it as mips-linux-musl.
>
> Then I went through the test model for vanadis found in vanadis/tests/basic_vanadis.py and noticed it was directed at the root folder of the hello-world file, rather than the file itself:
> basic_vanadis.py:131:
> "executable" : os.getenv("VANADIS_EXE", "./tests/small/basic-io/hello-world"),
> => "executable" : os.getenv("VANADIS_EXE", "./tests/small/basic-io/hello-world/mipsel/hello-world"),
>
> With that modified, my error has now changed from ELF issues, into segfaults - both on Ubuntu 20 as well as Fedora 36
> Ubuntu 20
> [image]<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F106702975%2F176730855-64734747-1d0d-4dc5-8ac7-ddd0ee78c69d.png&data=05%7C01%7Cjpkenny%40sandia.gov%7Cccaa2c73453040aa618408da5ab70c91%7C7ccb5a20a303498cb0c129007381b574%7C1%7C0%7C637922039480668812%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=yaoMoQwTY5PBue3bVXhDjyPu7aPgoQ68zlkfd85j3OI%3D&reserved=0>
> Fedora 36
> [image]<https://gcc02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F106702975%2F176731347-5965d623-ceb6-4ac0-8163-4fd3bb80ede3.png&data=05%7C01%7Cjpkenny%40sandia.gov%7Cccaa2c73453040aa618408da5ab70c91%7C7ccb5a20a303498cb0c129007381b574%7C1%7C0%7C637922039480824618%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C3000%7C%7C%7C&sdata=ERd5MeJZIYfbOxJm0GYwgpSaSo3wQqy1GmhTwZuky9g%3D&reserved=0>
>
> It may be an issue with basic_vanadis.py being outdated? Is there another way I can easily check vanadis to ensure it's functioning properly?

I think we’re both hitting the same memory bug now, which is… progress…

I’m on my phone now, so it’s hard to be verbose, but if you look at my comments on your issue and my recent core issue, you could, in principle, piece together how to use -fsanitize=address to verify the memory bug. But I’m pretty confident that’s what’s happening, so there’s not a ton of value in that exercise.

We’re in a bit of a bind as the primary Vanadis developer recently left Sandia. We have to sort out on our end how our team is going to move forward. I wish I had a better answer for you.

Joe

nmhamster, 2022-10-28:
Can you try using the recent updates to Vanadis that were merged (into `devel`) yesterday? These contain a major number of fixes for the pipeline code. I will close the issue but please re-open it if the problem re-occurs.

closed 2022-10-28 by nmhamster