sstsimulator/sst-elements #1914: SST Can't find a new component

prateek22sri, 2022-06-02:
On my mac at ~/Software/scratch/src/sst-elements-library-12.0.0/src/sst/elements/ location, I created the following files:
├── Makefile.am
├── hello.cc
├── hello.h
└── tests
    └── hello.py

Makefile.am 
`# -*- Makefile -*-
#
#

AM_CPPFLAGS += \
	$(MPI_CPPFLAGS)

compdir = $(pkglibdir)
comp_LTLIBRARIES = libhello.la

# sstdir = $(includedir)/sst/elements/hello

libhello_la_SOURCES = \
	hello.cc \
	hello.h

deprecated_libhello_sources =

if !SST_ENABLE_PREVIEW_BUILD
libhello_la_SOURCES += $(deprecated_libhello_sources)
endif

EXTRA_DIST = \
    README \
    tests/hello.py

deprecated_EXTRA_DIST =

if !SST_ENABLE_PREVIEW_BUILD
EXTRA_DIST += $(deprecated_EXTRA_DIST)
endif

libhello_la_LDFLAGS = -module -avoid-version

install-exec-hook:
	$(SST_REGISTER_TOOL) SST_ELEMENT_SOURCE     hello=$(abs_srcdir)
	$(SST_REGISTER_TOOL) SST_ELEMENT_TESTS      hello=$(abs_srcdir)/tests`

hello.py
`import sst

helloComponent = sst.Component("hello", "exA_helloSST.hello")
helloComponent.addParams({
    "repeats" : "5"
})`

hello.cc
`#include "sst_config.h"
#include "hello.h"

namespace exA_hello {
    hello::hello( SST::ComponentId_t id, SST::Params& params) : SST::Component(id), repeats(0) {

        // initialize output
        output.init("hello." + getName() + "->", 1, 0, SST::Output::STDOUT);

        // 10 is default, overwritten by project driver
        maxRepeats = params.find<SST::Cycle_t>("repeats", 10);

        // Show user the parameters
        output.verbose(CALL_INFO, 1, 0, "Config: repeats=%" PRIu64 "\n", static_cast<uint64_t>(maxRepeats));

[... 543 lines cut ...]
make install

```
[mac:~/Software/scratch/src/sst-elements-library-12.0.0] $ cd src/sst/elements/hello/tests 
[mac:~/Software/scratch/src/sst-elements-library-12.0.0/src/sst/elements/hello/tests] $ sst hello.py 
WARNING: Building component "hello" with no links assigned.
FATAL:  SST Core: can't find requested component 'exA_helloSST.hello'
Error: unable to find "exA_helloSST" element library

SST Fatal Backtrace Information:
    0 : 0   sstsim.x                            0x00000001040f9fa4 _ZNK3SST6Output5fatalEjPKcS2_iS2_z + 612
    1 : 1   sstsim.x                            0x00000001040ee508 _ZN3SST7Factory15CreateComponentEyRKNSt3__112basic_stringIcNS1_11char_traitsIcEENS1_9allocatorIcEEEERNS_6ParamsE + 476
    2 : 2   sstsim.x                            0x00000001041031bc _ZN3SST15Simulation_impl13performWireUpERNS_11ConfigGraphERKNS_8RankInfoEy + 124
    3 : 3   sstsim.x                            0x00000001040afd30 _ZL16start_simulationjR15SimThreadInfo_tRN3SST4Core10ThreadSafe7BarrierE + 788
    4 : 4   sstsim.x                            0x00000001040acb94 main + 7892
    5 : 5   dyld                                0x000000010478508c start + 520
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------
```

prateek22sri, 2022-06-04:
Fixed it.

closed 2022-06-04 by prateek22sri

rahahahat, 2022-06-16:
> Hi @rahahahat,
> 
> I have created two blog posts. They might help you.
> 
> 1. This lets you create custom component in your install folder. I don't recommend this technique:
>    `https://www.theprateeksrivastava.com/sst-simulator-creating-custom-component-within-source-tree/`
> 2. This lets you create custom component from any folder once you have set your paths correctly. I recommend this technique.
>    `https://www.theprateeksrivastava.com/sst-simulator-external-component/ `
> 
> Best, Prateek

Thanks alot. This is perfect