sstsimulator/sst-elements #1915: Using NIC between Vanadis LSQ and L1Dcache

William-An, 2022-06-02:
Hello all,

I am currently working on an MMIO device and want to integrate that with Vanadis. I intended to create an NIC between the Vanadis LSQ, L1dcache, and the mmio device so that writes and reads to it could bypass the cache. However, following similar configurations in `testStdMem-mmio.py`, the Vanadis CPU seems to be stuck at fetch or decode with no instructions being executed.

Attached is a launch script to reproduce the issue.

SST Launch script:
```python
import os
import sst

# Define SST core options
sst.setProgramOption("timebase", "1ps")
sst.setProgramOption("stopAtCycle", "0 ns")

# Tell SST what statistics handling we want
sst.setStatisticLoadLevel(4)

verbosity = int(os.getenv("VANADIS_VERBOSE", 0))
os_verbosity = os.getenv("VANADIS_OS_VERBOSE", verbosity)
pipe_trace_file = os.getenv("VANADIS_PIPE_TRACE", "")
lsq_entries = os.getenv("VANADIS_LSQ_ENTRIES", 32)

rob_slots = os.getenv("VANADIS_ROB_SLOTS", 64)
retires_per_cycle = os.getenv("VANADIS_RETIRES_PER_CYCLE", 4)
issues_per_cycle = os.getenv("VANADIS_ISSUES_PER_CYCLE", 4)
decodes_per_cycle = os.getenv("VANADIS_DECODES_PER_CYCLE", 4)

integer_arith_cycles = int(os.getenv("VANADIS_INTEGER_ARITH_CYCLES", 2))
integer_arith_units = int(os.getenv("VANADIS_INTEGER_ARITH_UNITS", 2))
fp_arith_cycles = int(os.getenv("VANADIS_FP_ARITH_CYCLES", 8))
fp_arith_units = int(os.getenv("VANADIS_FP_ARITH_UNITS", 2))
branch_arith_cycles = int(os.getenv("VANADIS_BRANCH_ARITH_CYCLES", 2))

auto_clock_sys = os.getenv("VANADIS_AUTO_CLOCK_SYSCALLS", "no")

cpu_clock = os.getenv("VANADIS_CPU_CLOCK", "2.3GHz")

vanadis_cpu_type = "vanadisdbg.VanadisCPU"

#if (verbosity > 0):
#	print("Verbosity (" + str(verbosity) + ") is non-zero, using debug version of Vanadis.")
#	vanadis_cpu_type = "vanadisdbg.VanadisCPU"

print("Verbosity: " + str(verbosity) + " -> loading Vanadis CPU type: " + vanadis_cpu_type)
print("Auto-clock syscalls: " + str(auto_clock_sys))

v_cpu_0 = sst.Component("v0", vanadis_cpu_type)
v_cpu_0.addParams({
       "clock" : cpu_clock,
#       "executable" : "./tests/hello-mips",
#       "executable" : "./tests/hello-musl",
#       "executable" : "./tests/core-perf-musl",
#       "executable" : "./tests/stream-musl",
#       "executable" : "./tests/stream-mini-musl",
#       "executable" : "./tests/test-printf",
#       "executable" : "./tests/test-env",
#       "executable" : "./tests/lulesh2.0",
#       "executable" : "./tests/luxtxt",
#       "executable" : "./tests/stream-fortran",
#       "executable" : "./tests/test-fp",
#       "executable" : os.getenv("VANADIS_EXE", "./tests/stream-mini-musl"),
       "executable" : os.getenv("VANADIS_EXE", "./vanadisHandshake/vanadisHandshake"),
       "app.env_count" : 2,
       "app.env0" : "HOME=/home/sdhammo",
       "app.env1" : "NEWHOME=/home/sdhammo2",
#       "app.argc" : 4,
#       "app.arg1" : "16",
#       "app.arg2" : "8",
#       "app.arg3" : "8",
[... 253 lines cut ...]
link_l1dcache_l2cache_link = sst.Link("link_l1dcache_l2cache_link")
link_l1dcache_l2cache_link.connect( (l1dcache_2_l2cache, "port", "1ns"), (cache_bus, "high_network_0", "1ns") )

link_l1icache_l2cache_link = sst.Link("link_l1icache_l2cache_link")
link_l1icache_l2cache_link.connect( (l1icache_2_l2cache, "port", "1ns"), (cache_bus, "high_network_1", "1ns") )

link_os_l1dcache_l2cache_link = sst.Link("link_os_l1dcache_l2cache_link")
link_os_l1dcache_l2cache_link.connect( (os_l1dcache, "low_network_0", "1ns"), (cache_bus, "high_network_2", "1ns") )

link_bus_l2cache_link = sst.Link("link_bus_l2cache_link")
link_bus_l2cache_link.connect( (cache_bus, "low_network_0", "1ns"), (l2cache_2_l1caches, "port", "1ns") )

link_l2cache_2_rtr = sst.Link("link_l2cache_2_rtr")
link_l2cache_2_rtr.connect( (l2cache_2_mem, "port", "1ns"), (comp_chiprtr, "port0", "1ns") )

link_dir_2_rtr = sst.Link("link_dir_2_rtr")
link_dir_2_rtr.connect( (comp_chiprtr, "port1", "1ns"), (dirNIC, "port", "1ns") )

link_dir_2_mem = sst.Link("link_dir_2_mem")
link_dir_2_mem.connect( (dirtoM, "port", "1ns"), (memToDir, "port", "1ns") )

link_core0_os_link = sst.Link("link_core0_os_link")
link_core0_os_link.connect( (os_hdlr, "os_link", "5ns"), (node_os, "core0", "5ns") )

```

hughes-c, 2022-06-06:
@William-An I don't see an mmio space in this configuration. Can you turn up the verbosity and provide a trace where it is hanging? What compiler are you using?

William-An, 2022-06-06:
I did not connect the MMIO device in this configuration to focus on the NIC connection part, as even without the MMIO device, the NIC connection between LSQ and the L1dcache still does not work. 

For the compiler, I compiled the vanadis binary with `mipsel-linux-musl-gcc (GCC) 10.2.1 20210227`. For the sst framework, I used GCC 4.9.2.

## Traces
### Start of traces
```
Verbosity: 20 -> loading Vanadis CPU type: vanadisdbg.VanadisCPU
Auto-clock syscalls: no
No application arguments found, continuing with argc=0
[Core:    0/0]: Executable: ./vanadisHandshake/vanadisHandshake
[Core:    0/0]: Reading Program Header 0...
[Core:    0/0]: Unknown program header type in ELF: 1879048195
[Core:    0/0]: Reading Program Header 1...
[Core:    0/0]: Unknown program header type in ELF: 1879048192
[Core:    0/0]: Reading Program Header 2...
[Core:    0/0]: Reading Program Header 3...
[Core:    0/0]: Reading Program Header 4...
[Core:    0/0]: Unknown program header type in ELF: 1685382481
[Core:    0/0]: Reading Section Header 0...
[Core:    0/0]: Reading Section Header 1...
[Core:    0/0]: Unknown Section type: 1879048234
[Core:    0/0]: Reading Section Header 2...
[Core:    0/0]: Unknown Section type: 1879048198
[Core:    0/0]: Reading Section Header 3...
[Core:    0/0]: Reading Section Header 4...
[Core:    0/0]: Reading Section Header 5...
[Core:    0/0]: Reading Section Header 6...
[Core:    0/0]: Reading Section Header 7...
[Core:    0/0]: Reading Section Header 8...
[Core:    0/0]: Reading Section Header 9...
[Core:    0/0]: Reading Section Header 10...
[Core:    0/0]: Reading Section Header 11...
[Core:    0/0]: Reading Section Header 12...
[Core:    0/0]: Reading Section Header 13...
[Core:    0/0]: Reading Section Header 14...
[Core:    0/0]: Reading Section Header 15...
[Core:    0/0]: Reading Section Header 16...
[Core:    0/0]: Reading Section Header 17...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 18...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 19...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 20...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 21...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 22...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 23...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 24...
[Core:    0/0]: Unknown Section type: 1879048222
[Core:    0/0]: Reading Section Header 25...
[Core:    0/0]: Unknown Section type: 1879048181
[Core:    0/0]: Reading Section Header 26...
[Core:    0/0]: Reading Section Header 27...
[Core:    0/0]: Reading Section Header 28...
[Core:    0/0]: Reading Section Header 29...
[Core:    0/0]: Reading in symbol table (Section 27)...
[Core:    0/0]: Symbol Table located at 65532 / 0xfffc in executable (is 32bit? yes).
[Core:    0/0]: Symbol entry is 16 bytes in size, expecting 225 symbols (remainder: 0)
[Core:    0/0]: Binary is marked as 32bit
[Core:    0/0]: Read of section completed.
[Core:    0/0]: -------------------------------------------------------
[Core:    0/0]: ELF Information [./vanadisHandshake/vanadisHandshake]
[Core:    0/0]: -------------------------------------------------------
[Core:    0/0]: Class:                32-bit
[Core:    0/0]: Instruction-Set:      8 / MIPS
[... 2501 lines cut ...]
[Core:    0/975705]: | isa:    0 -> phys:  127 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    1 -> phys:  126 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    2 -> phys:  125 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    3 -> phys:  124 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    4 -> phys:  123 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    5 -> phys:  122 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    6 -> phys:  121 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    7 -> phys:  120 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    8 -> phys:  119 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:    9 -> phys:  118 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   10 -> phys:  117 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   11 -> phys:  116 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   12 -> phys:  115 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   13 -> phys:  114 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   14 -> phys:  113 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   15 -> phys:  112 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   16 -> phys:  111 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   17 -> phys:  110 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   18 -> phys:  109 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[... cut ...] :    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   20 -> phys:  107 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   21 -> phys:  106 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   22 -> phys:  105 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   23 -> phys:  104 | r:    0 | w:    0 | v: 0x0000000000000000 | v: 0 / 0
[Core:    0/975705]: | isa:   24 -> phys:  103 | r

hughes-c, 2022-06-08:
What's the hash of the version that you're using? Your configuration is no longer valid with the master/devel branch of sst-elements.

William-An, 2022-06-08:
The commit hash is `bfbdc5c4865518263856337625d567e4afc7604c`

nmhamster, 2022-07-08:
@gvoskuilen - can you confirm that the standard memory interface (at the component side) can function with components in between this and the first level cache? I remember way back that this situation was complex as the cache converted the request into commands for the memory subsystem. Just want to check that we are expecting this to work or not work (which ever!).

gvoskuilen, 2022-07-08:
@nmhamster Yes, this works now to have a network between the LSQ & L1. So theoretically the config should work, I'm looking at it now.

William-An, 2022-08-11:
Hello all, just an update on this issue.

I updated both the sst core and sst elements to the newest main branches and the above configuration file works now.

closed 2022-08-11 by William-An