sstsimulator/sst-elements #1920: Vanadis attempts to access freed instruction (memory bug)

jpkenny, 2022-06-29:
-fsanitize=address flags a memory bug in Vanadis (12.0.0 release) where a previously freed instruction is accessed. With the right compiler and environment all kinds of nonsense ensue.

```
Verbosity: 0 -> loading Vanadis CPU type: vanadisdbg.VanadisCPU
Auto-clock syscalls: no
No application arguments found, continuing with argc=0
WARNING: Element vanadisdbg.VanadisCPU has slot with an invalid name: decoder%(hardware_threads)d
l2cache: No MSHR lookup latency provided (mshr_latency_cycles)...intrapolated to 3 cycles.
memory, WARNING: Memories no longer inherit address regions from directories and no region parameters (addr_range_start, addr_range_end, interleave_size, interleave_step) were detected. All addresses will map to this memory: if this is intended, you may ignore this warning or set addr_range_start to 0 in your input deck to eliminate this warning.
=================================================================
==321474==ERROR: AddressSanitizer: heap-use-after-free on address 0x60d000003ac8 at pc 0x7f6ceefa930f bp 0x7ffe3c45e960 sp 0x7ffe3c45e950
READ of size 8 at 0x60d000003ac8 thread T0
    #0 0x7f6ceefa930e in SST::Vanadis::VanadisInstruction::getInstructionAddress() const (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x32a30e)
    #1 0x7f6ceefd2b91 in SST::Vanadis::VanadisSequentialLoadStoreQueue::StandardMemHandlers::handle(SST::Interfaces::StandardMem::WriteResp*) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x353b91)
    #2 0x7f6ceefbd4ee in SST::Interfaces::StandardMem::WriteResp::handle(SST::Interfaces::StandardMem::RequestHandler*) /home/jpkenny/install/sstcore-12.0.0-gcc10/include/sst/core/interfaces/stdMem.h:440
    #3 0x7f6ceefd3a98 in SST::Vanadis::VanadisSequentialLoadStoreQueue::processIncomingDataCacheEvent(SST::Interfaces::StandardMem::Request*) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x354a98)
    #4 0x7f6cef19cde4 in SST::SSTHandler<void, SST::Interfaces::StandardMem::Request*, false, SST::Vanadis::VanadisSequentialLoadStoreQueue, void>::operator()(SST::Interfaces::StandardMem::Request*) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x51dde4)
    #5 0x7f6cee5ca0d8 in SST::MemHierarchy::StandardInterface::receive(SST::Event*) /home/jpkenny/src/sst-elements-library-12.0.0-gcc10/src/sst/elements/memHierarchy/standardInterface.cc:308
    #6 0x7f6cee5f2c9e in SST::SSTHandler<void, SST::Event*, false, SST::MemHierarchy::StandardInterface, void>::operator()(SST::Event*) /home/jpkenny/install/sstcore-12.0.0-gcc10/include/sst/core/ssthandler.h:161
    #7 0x7f6cee4a5eac in SST::MemHierarchy::MemLinkBase::recvNotify(SST::Event*) ../../../../src/sst/elements/memHierarchy/memLinkBase.h:159
    #8 0x7f6cee4b004a in SST::SSTHandler<void, SST::Event*, false, SST::MemHierarchy::MemLink, void>::operator()(SST::Event*) /home/jpkenny/install/sstcore-12.0.0-gcc10/include/sst/core/ssthandler.h:161
    #9 0x5647b27fd400 in SST::Event::execute() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/event.cc:60
    #10 0x5647b282e0b5 in SST::Simulation_impl::run() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/simulation.cc:673
    #11 0x5647b2755ad0 in start_simulation /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:424
    #12 0x5647b275708f in main /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:817
    #13 0x7f6cf36dcd8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #14 0x7f6cf36dce3f in __libc_start_main_impl ../csu/libc-start.c:392
    #15 0x5647b2754394 in _start (/home/jpkenny/install/sstcore-12.0.0-gcc10/libexec/sstsim.x+0x425394)

0x60d000003ac8 is located 8 bytes inside of 144-byte region [0x60d000003ac0,0x60d000003b50)
freed by thread T0 here:
    #0 0x7f6cf42f522f in operator delete(void*, unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:172
    #1 0x7f6ceefea008 in SST::Vanadis::VanadisStoreInstruction::~VanadisStoreInstruction() (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x36b008)
    #2 0x7f6ceef93fc4 in SST::Vanadis::VanadisDebugComponent::performRetire(SST::Vanadis::VanadisCircularQueue<SST::Vanadis::VanadisInstruction*>*, unsigned long) /home/jpkenny/src/sst-elements-library-12.0.0-gcc10/src/sst/elements/vanadis/vanadis.cc:1000
    #3 0x7f6ceef96594 in SST::Vanadis::VanadisDebugComponent::tick(unsigned long) /home/jpkenny/src/sst-elements-library-12.0.0-gcc10/src/sst/elements/vanadis/vanadis.cc:1293
    #4 0x7f6cef15d868 in SST::SSTHandler<bool, unsigned long, true, SST::Vanadis::VanadisDebugComponent, void>::operator()(unsigned long) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x4de868)
    #5 0x5647b2790b4f in SST::Clock::execute() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/clock.cc:113
    #6 0x5647b282e0b5 in SST::Simulation_impl::run() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/simulation.cc:673
    #7 0x5647b2755ad0 in start_simulation /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:424
    #8 0x5647b275708f in main /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:817
    #9 0x7f6cf36dcd8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

previously allocated by thread T0 here:
    #0 0x7f6cf42f41c7 in operator new(unsigned long) ../../../../src/libsanitizer/asan/asan_new_delete.cpp:99
    #1 0x7f6ceefaf622 in SST::Vanadis::VanadisStoreInstruction::clone() (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x330622)
    #2 0x7f6ceeff9832 in SST::Vanadis::VanadisMIPSDecoder::tick(SST::Output*, unsigned long) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x37a832)
    #3 0x7f6ceef8f2fe in SST::Vanadis::VanadisDebugComponent::performDecode(unsigned long) /home/jpkenny/src/sst-elements-library-12.0.0-gcc10/src/sst/elements/vanadis/vanadis.cc:526
    #4 0x7f6ceef95fba in SST::Vanadis::VanadisDebugComponent::tick(unsigned long) /home/jpkenny/src/sst-elements-library-12.0.0-gcc10/src/sst/elements/vanadis/vanadis.cc:1246
    #5 0x7f6cef15d868 in SST::SSTHandler<bool, unsigned long, true, SST::Vanadis::VanadisDebugComponent, void>::operator()(unsigned long) (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x4de868)
    #6 0x5647b2790b4f in SST::Clock::execute() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/clock.cc:113
    #7 0x5647b282e0b5 in SST::Simulation_impl::run() /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/simulation.cc:673
    #8 0x5647b2755ad0 in start_simulation /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:424
    #9 0x5647b275708f in main /home/jpkenny/src/sstcore-12.0.0-gcc10/src/sst/core/main.cc:817
    #10 0x7f6cf36dcd8f in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58

SUMMARY: AddressSanitizer: heap-use-after-free (/home/jpkenny/install/sst-elements-library-12.0.0-gcc10/lib/sst-elements-library/libvanadis.so+0x32a30e) in SST::Vanadis::VanadisInstruction::getInstructionAddress() const
Shadow bytes around the buggy address:
  0x0c1a7fff8700: 00 00 fa fa fa fa fa fa fa fa fd fd fd fd fd fd
  0x0c1a7fff8710: fd fd fd fd fd fd fd fd fd fd fd fd fa fa fa fa
  0x0c1a7fff8720: fa fa fa fa 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1a7fff8730: 00 00 00 00 00 00 fa fa fa fa fa fa fa fa 00 00
  0x0c1a7fff8740: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c1a7fff8750: fa fa fa fa fa fa fa fa fd[fd]fd fd fd fd fd fd
  0x0c1a7fff8760: fd fd fd fd fd fd fd fd fd fd fa fa fa fa fa fa
  0x0c1a7fff8770: fa fa 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c1a7fff8780: 00 00 00 00 fa fa fa fa fa fa fa fa 00 00 00 00
  0x0c1a7fff8790: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 fa fa
  0x0c1a7fff87a0: fa fa fa fa fa fa fd fd fd fd fd fd fd fd fd fd
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
  Shadow gap:              cc
==321474==ABORTING
```

jpkenny, 2022-06-29:
Verified same memory bug appears in devel

nmhamster, 2022-07-08:
@jpkenny - this is a known bug and was introduced when the LSQ was overhauled to use the new standard-mem interface. A fix is on the way for this specific issue.

ErsinCK, 2022-10-20:
is there any update on this issue? I wanted to use vanadis for my project, but it looks like I am running into this issue. In my case, it shows up as "cannot read ELF file" when I try to run the simple tests..

nmhamster, 2022-10-28:
Fairly significant number of changes to Vanadis were merged yesterday. Please try the `devel` branch to obtain the code changes. Will close but re-open if the issue re-occurs.

closed 2022-10-28 by nmhamster