sstsimulator/sst-elements #1925: Vanadis Segfault

Connie120, 2022-07-15:
Hello! I downloaded the latest sst elements and sst core from the master branch and tried to run the hello-world example in the tests/basic-io/hello-world/riscv64/ folder. But I encountered this segfault, and here's the backtrace info. 

[image]

The command I used to run is "sst tests/basic_vanadis.py" and I've changed "MIPS" to "RISCV64". The operating system I'm using is Ubuntu 20.04. Any help would be appreciated, thank you!

jpkenny, 2022-07-18:
Hi,
This is likely a duplicate of issue #1920. There is a known bug in Vanadis and work on a fix is underway.

nmhamster, 2022-10-28:
Will close this issue, please re-open if this problem re-occurs.

closed 2022-10-28 by nmhamster