sstsimulator/sst-elements #1929: uint is not a type

pdbj, 2022-08-05:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
GNA and CramSim use the type `uint`, which is not a standard type.

2 - Describe how to reproduce
Compile sst-elements and it will fail

One example
```
Making all in CramSim
sst-elements/src/sst/elements/CramSim
  CXX      c_TxnGen.lo
c_TxnGen.cpp: In constructor 'SST::CramSim::c_TxnGenBase::c_TxnGenBase(SST::ComponentId_t, SST::Params&)':
c_TxnGen.cpp:78:21: error: 'uint' was not declared in this scope; did you mean 'rint'?
   78 |     m_sizeOffset = (uint)log2(k_numBytesPerTransaction);
      |                     ^~~~
      |                     rint

```

3 - What Operating system(s) and versions 
gcc-10.2.1
Mac

4 - What version of external libraries (Boost, MPI)
NA

5 - Provide sha1 of all relevant sst repositories (sst-core, sst-elements, etc)
sst/devel @ bcd759f8855943c2
sst-elements/devel @ https://github.com/sstsimulator/sst-elements/commit/19a6ad660b6d44dffc605f0981b33154ecbd7419

6 - Changing all these to `unsigned` at least fixes the compile.

pdbj, 2022-08-05:
[patch-uint.diff.txt](https://github.com/sstsimulator/sst-elements/files/9273847/patch-uint.diff.txt)

pdbj, 2022-08-05:
Missed one:

```
diff --git a/src/sst/elements/GNA/sts.h b/src/sst/elements/GNA/sts.h
index 9938753f2..37e4f4013 100644
--- a/src/sst/elements/GNA/sts.h
+++ b/src/sst/elements/GNA/sts.h
@@ -44,7 +44,7 @@ public:
     STS(GNA *parent, int n) : myGNA(parent), stsID(n), numSpikes(0) {;}
     bool isFree();
     void assign(int);
-    void advance(uint);
+    void advance(unsigned);
     void returnRequest(SST::Interfaces::StandardMem::Request *req) {
         incomingReqs.push(req);
     }
```

nmhamster, 2022-11-21:
This this issue is closed from #1955, please re-open if you have issues.

closed 2022-11-21 by nmhamster