sstsimulator/sst-elements #1935: Filter control packets (Merlin)

tommasobo, 2022-09-20:
Hello,

I was analyzing the network traffic in a Merlin router and observed that at the beginning of each benchmark there seems to be some control packets which are sent in the network (potentially I observed also some in the middle or at the end of the benchmark).

Is there any easy way to filter out these packets? One very quick way I found is to filter them by packet size but it is obviously not ideal. I tried understanding where they are created so I could potentially add a flag to those FireFly packets, but I couldn't find the exact file.

Do you have any suggestion about how to filter them or how to mark these packets? (basically all the packets that are not created directly by a MPI call in ember)
Thanks!

feldergast, 2022-09-20:
I would need to know your configuration to be able to give a complete answer.  I can say, that there are control event that flow between the PortControl and LinkControl blocks during the init() phase of simulation to set up bandwidths, credits, virtual networks, etc.  For the most part, after init() and setup() all the packets being sent would likely be from ember (unless you enabled congestion management, which can send control messages during runtime).

tommasobo, 2022-09-21:
Hi @feldergast , our configuration is very straightforward. We only edited some topology files (dragonfly.cc and fattree.cc) to add some new routing algorithms and then we do some data collection and analysis in hr_router.cc and nicRecvMachine.cc

We launch all our benchmarks from ember using the supported mpi collectives (enQ_send() etc..).

Yes, we were wondering if there is an easy method to flag those init() / setup() packets or to filter them in the routers / NICs. Thanks!

feldergast, 2022-09-21:
What are you trying to filter them from?  Where are you data collection points in the code?  Anything send during init() and complete() should be taking different paths through the code than a regular event would.

If you're already instrumenting the code, you could add a variable to the proper classes to track when you are in the various stages of the simulation and then ignore things during the stages you're not interested in.

closed 2023-10-24 by hughes-c