sstsimulator/sst-elements #1937: Ember/Merlin: Questions regarding multiple NICs per node and intranode communication

saichennaintel, 2022-10-05:
Hi,

I'm using Ember & Merlin for HPC network simulations. I have couple of questions regarding the _numCores_ and _nicsPerNode_ parameters while building an Ember Job (I'm concerned about EmberMPIJob currently). Before pointing out the issue I'm having, let me point down the couple of assumptions I had (please correct them as you see fit).

Assumption 1: _numCores_ argument provided in the EmberJob constructor defines the # of cores **per Node** (not total number of cores in the job)
Assumption 2: _nicsPerNode_ argument provided in the EmberJob constructor defines the # of NICs assigned per Node. 
Assumption 3: k = int(numCores/nicsPerNode) defines the number of cores connected per virtual NIC (Looks like each node has one NIC and k virtual NICs)

Issue : To understand how intranode and internode communication is simulated, I was trying to run a simple test configuration file (attached); a single router topology with 2 nodes (4 cores per node, 2 nics per node) , Ember _AllReduce_ motif being used. However, the simulation throws an error. The error does not arise when I use 1 nic per node. Upon debugging I found out that it is an **_out of index_** error. While _self._available_nodes_ (in pymerlin-base.py) shows all the nodes (i.e., [0,1]), during allocation it is asking for first _num_nodes_ (pymerlin_base.py: line 877). _num_nodes_ is calculated based on the _job.size_ parameter which has been initialized to _numNodes*nicsPerNode_ (pyember.py: line 25). As we are trying to access the list of size (_numNodes*nicsPerNode_) from a (_numNodes_)-sized list, I'm getting the out of index error. 

Is it a bug in the code or am I using initializing the parameters incorrectly? I see that the multi-NIC support has been added fairly recently (https://github.com/sstsimulator/sst-elements/pull/1411). 
[alltoall_singlerouter.txt](https://github.com/sstsimulator/sst-elements/files/9719516/alltoall_singlerouter.txt)

It would be really useful if you can describe little more on how the communication differs while using single NIC per node and multiple NICs per node. 
Also, can you elaborate on how **intranode communication** is being simulated. I was trying to look into the firefly module and _loopback.cc_, but it is a little overwhelming.  Considering a two-node single-router system described in the configuration file, i.e., 2 node system  with 4 cores on each node, what path does an _MPISend_ between rank0 and rank1 take (assuming linear mapping)? Also what parameters are considered while calculating the transfer time during such intra-node communication?

feldergast, 2022-10-06:
There are a couple issues going on here:

1 - You need to specify the total number of network ports in your topology.  In this case, you would need 4 ports (2 nodes and 2 NICs per node). So:
    topo.num_ports = 4

2 - You need to tell the System object to allocate the endpoints in sets of 2 ports since there are two NICs per node.  So:
    system.setTopology(topo,2)

With those two changes, I think things should work for you.

I don't know the details of intranode communication since I didn't write that part of the code, but I do know it is trying to model a shared memory transport for messages that are going between ranks on the same node.

One more thing to node, numCores, really refers to the number of MPI ranks.  Ember does not try to model threads..

saichennaintel, 2022-10-07:
Hello @feldergast, thank you for the quick response.
1.	Thanks for pointing out the _topo.num_ports_ error. I missed that and it makes total sense.
2.	Second fix also makes sense. I have updated that as well. 

But I’m mainly confused about the _num_nodes_ argument passed to an EmberMPIJob() ([pyember.py](https://github.com/sstsimulator/sst-elements/blob/master/src/sst/elements/ember/pyember.py): line 189). What does that parameter/argument mean? Does it mean the total number of MPI ranks requested in
this job or total number of endpoints (also what represents an endpoint? Does each MPI rank represent an endpoint?)

I’m trying to simulate a single router system with 4 ports. Two nodes are attached to that single router. Each node has 2 NICs which are attached to the two ports of the router. Additionally, each node has 4 cores. I have an attached an image below visually representing the system mentioned above. Now, in order to create an MPI Job where each MPI rank is bound to a core, I use the following function call EmberMPIJob(job_id=0,num_nodes=2,numCores=4,nicsPerNode=2). Config file attached. I get the same error I was pointing out earlier. 

[image]
[allreduce_singlerouter_modified.txt](https://github.com/sstsimulator/sst-elements/files/9737836/allreduce_singlerouter_modified.txt)

So how should I initialize the EmberMPIJob() in order to ensure that I run a motif with 8 MPI ranks (=2nodes*4cores-each) for the above simulated system, i.e., a 2 node system with 4 cores each having 2 NICs per node?

One last question. The last line in your previous comment is confusing. Did you mean "One more thing to **note**, numCores, really refers to the number of MPI ranks. Ember does not try to model threads." ? Or did you mean "One more thing, _node_, _numCores_, really refers to the number of MPI ranks. Ember does not try to model threads."? If you meant the latter, how to calculate total number of MPI ranks for the job by looking at _num_nodes_ and _numCores_ parameters?

I apologise in advance for a long series of questions. Just trying to understand the tool better to make better sense of the simulations.

feldergast, 2022-10-10:
Your test file runs successfully for me, so I can't reproduce the error to debug it.  I can give a little more detail in how things work:

The EmberMPIJob constructor takes:
1 - Job ID - doesn't matter what it is, as long it is unique for each job. We make the user specify it so that messages with the job id can be mapped back to the right job
2 - Number of nodes (num_nodes) - this is actually the number of nodes, not the number of endpoints
3 - Number of cores (numCores) - this is really just the number of MPI ranks per node.  Total ranks in the job will be num_nodes * numCores.
4 - Number of nics per node (nicsPerNode) - number of nics per node.  The job will use num_nodes * nicsPerNode endpoints on the network.  Also, the numCores must be an even multiple of nicsPerNode and each "core" is mapped to a single NIC (i.e., a single core can't use the bandwidth of multiple nics).

So, the EmberMPIJob in the sample file is initialized correctly.  I'm not sure why you're seeing an error and I'm not. Which version of the code are you using?  I was testing on the head of the devel branch, but none of the pertinent files have been changed since the SST 12 release.

saichennaintel, 2022-10-14:
Thank you for the explanation on the EmberMPIJob constructor arguments. It clarifies my doubts. Regarding the error I had, I have modified the hr_router class in pymerlin-base.py of SST version 11.0.0, which could be the culprit. I have tested the above configuration file with a new installation of SST 12.0.1 and it worked fine.

closed 2022-10-14 by saichennaintel