sstsimulator/sst-elements #1953: Mercury element fails to compile on ubuntu 18.04 with GCC 7.50

jleidel, 2022-10-20:
Ubuntu 18.04 LTS; x86_64; gcc (Ubuntu 7.5.0-3ubuntu1~18.04) 7.5.0 (installed from apt)

The mercury element fails to compile from the devel branch.  The last commit on mercury was from @jpkenny.  The linker appears to not like the existence of `__exit` symbols in the binary.  This may be a configuration issue as the linker complains that the objects are not built with `-fPIC`.  

Note that this *works* on Ubuntu 21.04 with gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0.  

```
/usr/bin/ld: operating_system/threading/asm/.libs/make_x86_64_sysv_macho_gas.o: relocation R_X86_64_PC32 against undefined symbol `__exit' can not be used when making a shared object; recompile with -fPIC
```

jpkenny, 2022-10-21:
Thanks for the report. In theory libtool should be handling -fPIC, so I'll have to dig into what's going on here.

jpkenny, 2022-10-24:
Definitely a bug, but the PIC warning is a red herring.
Just needed to fix asm handling in the makefile.
PR #1954 should resolve issue.

jleidel, 2022-12-20:
fixed upstream