sstsimulator/sst-elements #1963: Merlin: Library/Module not found upon adding a new topology

saichennaintel, 2022-11-07:
I'm adding a new topology into SST Merlin library. In order to add this new topology, I have implemented a python build file which instantiates the routers and establishes links across the routers and endpoints. I was able to verify the correctness of my python file (pymerlin-topo-topology1.py) by dumping the output configuration graph using --output-dot argument. I have also implemented the routing logic in topology1.cc & topology1.h files. Following is the file structure representing the actual implementation:

/merlin/topology: --> pymerlin-topo-topology1.py
/merlin/topology: -->topology1.cc
/merlin/topology: -->topology1.h
/merlin/topology: -->helper.h (which is included in topology1.h)

In order to build Merlin with the following changes, I make the following changes to the Makefile.am and merlin.cc
In Makefile.am(changes are in bold):
......
......
topology/mesh.cc
topology/fattree.h
topology/fattree.cc
topology/dragonfly.h
topology/dragonfly.cc
topology/singlerouter.h
topology/singlerouter.cc
topology/hyperx.h
topology/hyperx.cc
**topology/topology1.h
topology/topology1.cc
topology/helper.h**

......
......
topology/pymerlin-topo-dragonfly.py
topology/pymerlin-topo-hyperx.py
topology/pymerlin-topo-fattree.py
topology/pymerlin-topo-mesh.py
**topology/pymerlin-topo-topology1.py**
......
......
BUILT_SOURCES = \
        pymerlin.inc \
        pymerlin-base.inc \
        pymerlin-endpoint.inc \
        pymerlin-router.inc \
        interfaces/pymerlin-interface.inc \
        target_generator/pymerlin-targetgen.inc \
        topology/pymerlin-topo-dragonfly.inc \
        ***topology/pymerlin-topo-topology1.inc \***
        topology/pymerlin-topo-hyperx.inc \
        topology/pymerlin-topo-fattree.inc \
        topology/pymerlin-topo-mesh.inc

In merlin.cc (changes in bold):

char pymerlin_topo_mesh[] = {
#include "topology/pymerlin-topo-mesh.inc"
0x00};

**char pymerlin_topo_topology1[] = {
#include "topology/pymerlin-topo-topology1.inc"
0x00};**
........
........

primary_module->addSubModule("topology",pymerlin_topo_fattree,"topology/pymerlin-topo-fattree.py");
primary_module->addSubModule("topology",pymerlin_topo_mesh,"topology/pymerlin-topo-mesh.py");
**primary_module->addSubModule("topology",pymerlin_topo_topology1,"topology/pymerlin-topo-topology1.py");**
........
........

Finally, once these modifications are made. I run the following commands:
 SST_ELEMENTS_ROOT:$ ./autogen.sh
 SST_ELEMENTS_ROOT:$ ./configure --prefix=$SST_ELEMENTS_HOME --with-sst-core=$SST_CORE_HOME
  SST_ELEMENTS_ROOT:$ make all && make install

Build process executed successfully with no errors. However, when running a sample test configuration file (An Ember pingpong motif), I get the "ModuleNotFound" error: (image attached)

[image]

Upon debugging I found out that the Merlin library has not been linked/registered. I get the following output when I do "sst-info merlin". Other elements do not show this error.

[image]

However, I see the libmerlin.so and libmerlin.la files in the above path, i.e., $SST_ELEMENTS_HOME/lib/sst-elements-library.

Am I missing any steps while rebuilding the Merlin library? Any help to fix this issue is highly appreciated.

feldergast, 2022-11-07:
I don't see anything wrong with the changes to the build system, so the most likely cause of this is a missing symbol in the element library (i.e. something defined in a header file, but not instantiated in a source file, for me it's usually an unimplemented destructor, but could be any function).  The sst-info error means that the file the library is contained in was found correctly, but the library was somehow illformed. We've fought for a long time to try to get better error reporting in these situations, but haven't struck on the right method yet for getting the actual errors to report.  I would go through and make sure that you don't have any missing symbols in your new classes as a first check.

saichennaintel, 2022-11-08:
Thank you @feldergast for pointing out the usual suspects to look at. Indeed the culprit was the improper implementation of destructor method. Was able to fix the issue. Thank you for the help.

closed 2022-11-08 by saichennaintel