sstsimulator/sst-elements #2003: merlin portcontrol has erroneous fields that induce buffer overflow

jleidel, 2022-12-20:
the merlin.portcontrol infrastructure induces buffer overflows when aborting.  during initialization, the merlin.portcontrol attempts to send initialization packets (regardless of what the top-level configuration is doing).  If these generate errors, then the abort message induces a buffer overflow. 

tested on SST-ELEMENTS-12.1.0; merlin.hr_router; merlin.singlerouter

*** buffer overflow detected ***: sst terminated
[pennywise:63969] *** Process received signal ***
[pennywise:63969] Signal: Aborted (6)
[pennywise:63969] Signal code:  (-6)
[pennywise:63969] [ 0] /lib/x86_64-linux-gnu/libpthread.so.0(+0x12980)[0x155554a72980]
[pennywise:63969] [ 1] /lib/x86_64-linux-gnu/libc.so.6(gsignal+0xc7)[0x1555537adfb7]
[pennywise:63969] [ 2] /lib/x86_64-linux-gnu/libc.so.6(abort+0x141)[0x1555537af921]
[pennywise:63969] [ 3] /lib/x86_64-linux-gnu/libc.so.6(+0x89967)[0x1555537f8967]
[pennywise:63969] [ 4] /lib/x86_64-linux-gnu/libc.so.6(+0x134b8f)[0x1555538a3b8f]
[pennywise:63969] [ 5] /lib/x86_64-linux-gnu/libc.so.6(+0x134bb1)[0x1555538a3bb1]
[pennywise:63969] [ 6] /lib/x86_64-linux-gnu/libc.so.6(+0x1328a0)[0x1555538a18a0]
[pennywise:63969] [ 7] /lib/x86_64-linux-gnu/libc.so.6(__vsnprintf_chk+0x105)[0x1555538a1055]
[pennywise:63969] [ 8] sst(_ZN3SST14vformat_stringB5cxx11EPKcP13__va_list_tag+0xae)[0x5555556ca0ae]
[pennywise:63969] [ 9] sst(_ZN3SST13format_stringB5cxx11EPKcz+0x9b)[0x5555556ca3fb]
[pennywise:63969] [10] sst(_ZNK3SST13BaseComponent6vfatalEjPKcS2_iS2_P13__va_list_tag+0x30f)[0x5555556573ff]
[pennywise:63969] [11] sst(_ZNK3SST13BaseComponent10sst_assertEbjPKcS2_iS2_z+0xa3)[0x5555556577d3]
[pennywise:63969] [12] /scratch/sst/sst-12.1.0/elements/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin11PortControl17checkInitProtocolEPNS_5EventENS0_12RtrInitEvent8CommandsEjPKcS7_+0x3d)[0x155541e5211d]
[pennywise:63969] [13] /scratch/sst/sst-12.1.0/elements/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin11PortControl4initEj+0x80)[0x155541e528f0]
[pennywise:63969] [14] /scratch/sst/sst-12.1.0/elements/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin9hr_router4initEj+0x660)[0x155541e25370]
[pennywise:63969] [15] sst(_ZN3SST15Simulation_impl10initializeEv+0xb4)[0x5555556c0a54]
[pennywise:63969] [16] sst(+0xe81b1)[0x55555563c1b1]
[pennywise:63969] [17] sst(main+0xb61)[0x55555561cec1]
[pennywise:63969] [18] /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xe7)[0x155553790bf7]
[pennywise:63969] [19] sst(_start+0x2a)[0x55555563b2fa]
[pennywise:63969] *** End of error message ***
Aborted (core dumped)

kaustubhcs, 2023-01-17:
Is there a mechanism to debug these errors? Any flags that could be set to give a traceback, pointing in the direction of the erroneous subcomponent?

jleidel, 2023-01-17:
> 

If you enable debugging, you may get more trace data.  The trace in the original issue contains the method where the abort occurs.  It comes from `Merlin::hr_router::init` [/scratch/sst/sst-12.1.0/elements/lib/sst-elements-library/libmerlin.so(_ZN3SST6Merlin9hr_router4initEj+0x660)[0x155541e25370]].

feldergast, 2023-01-17:
I can't reproduce this error.  Can you give me a minimal reproducer for it?  I've traced the code as best I could using the stack trace, which lead to the sst_assert() function.  This function uses vsnprintf with a limit of 256 bytes first and allocates a new buffer if that's not big enough and call it again.  There shouldn't be a way for that code to buffer overflow, unless there was an encoding error in vsnprintf(), which isn't checked for, and could cause a negative length to be passed into the second vsnprintf() call. Without a reproducer, I can't chase it any further.

feldergast, 2023-01-17:
> Is there a mechanism to debug these errors? Any flags that could be set to give a traceback, pointing in the direction of the erroneous subcomponent?

I there is no buffer overflow, the error message will print out the SubComponent that caused the error.

feldergast, 2023-01-17:
I found the typo that was causing the buffer overflow.  I have a fix in my current working branch and it will be in the next pull request I put in.