sstsimulator/sst-elements #2012: Ubuntu test elements failure

prateek22sri, 2023-01-16:
Ran 479 tests in 1151.221s

```
FAILED (failures=11, skipped=19)

=== TEST RESULTS BREAKDOWN ===========================================

testsuite_default_Messier.testcase_Messier_Component
 - FAILED  : test_Messier_gupsgen (testsuite_default_Messier.testcase_Messier_Component)
 - FAILED  : test_Messier_gupsgen_2RANKS (testsuite_default_Messier.testcase_Messier_Component)
 - FAILED  : test_Messier_gupsgen_fastNVM (testsuite_default_Messier.testcase_Messier_Component)
 - FAILED  : test_Messier_stencil3dbench_messier (testsuite_default_Messier.testcase_Messier_Component)
 - FAILED  : test_Messier_streambench_messier (testsuite_default_Messier.testcase_Messier_Component)

testsuite_default_Samba.testcase_Samba_Component
 - FAILED  : test_Samba_gupsgen_mmu (testsuite_default_Samba.testcase_Samba_Component)
 - FAILED  : test_Samba_gupsgen_mmu_4KB (testsuite_default_Samba.testcase_Samba_Component)
 - FAILED  : test_Samba_gupsgen_mmu_three_levels (testsuite_default_Samba.testcase_Samba_Component)
 - FAILED  : test_Samba_stencil3dbench_mmu (testsuite_default_Samba.testcase_Samba_Component)
 - FAILED  : test_Samba_streambench_mmu (testsuite_default_Samba.testcase_Samba_Component)

testsuite_default_cacheTracer.testcase_cacheTracer_Component
 - FAILED  : test_cacheTracer_1 (testsuite_default_cacheTracer.testcase_cacheTracer_Component)

=== TEST RESULTS SUMMARY =============================================

Tests Run            = 479
----------------------------------------
Passing              = 449
Failures             = 11
Skipped              = 19
Errors               = 0
Expected Failures    = 0
Unexpected Successes = 0
Testing Notes        = 0
----------------------------------------------------------------------
-- Total Test Time = 0 Hours, 19 Minutes, 11.222 Seconds --

====================
== TESTING FAILED ==
====================
```

prateek22sri, 2023-01-16:
The above message was received when running sst-test-elements. The sst-test-core ran fine. 
Does this mean I did not install SST elements correctly or is there an issue on SST side?

nhatdangncsu, 2023-02-16:
Did you figure out what the problem is?

prateek22sri, 2023-02-16:
No, haven't yet :(

feldergast, 2023-02-16:
Above the summary of failed tests in the output, you should be able to find the reasons that each test failed.  If you can post those, we may be able to help narrow down what's going on.

berquist, 2023-07-17:
@prateek22sri Do you have any new information?  Which version of Ubuntu?  What is your `configure.log`?