sstsimulator/sst-elements #2016: Hierarchical Topology Support for Merlin

Vatsanshuman, 2023-01-24:
Hi all,

I'm new to using SST and I'm trying to build a hierarchical topology, e.g. Ring+Mesh with Merlin network.
I was wondering if there's support present for building hierarchical topologies.

I referred the example for multiple networks: [Example 9 - SST tutorials](https://github.com/researcherben/structural-simulation-toolkit/tree/master/sst-tutorial-exercises/ex9)
I'm not sure if we can use the component "Bridge" to create hierarchical topology between Ring and Mesh.

I also checked the SST slides presented at ISCA 2017. [ISCA 2017](https://www.osti.gov/servlets/purl/1506211).
It doesn't mention any support for hierarchical topology.
[image]

If anyone has worked on hierarchical topologies, can you please help me with a reference of the same?

Thank you!

feldergast, 2023-01-25:
There is no direct support for hierarchical networks in merlin.  I doubt the bridge component will give you want you want, though it may be possible to make it functionally correct. The problem is that it probably couldn't be made to route on the shortest path for all src/dest pairs.  If you want a new hierarchical topology, it would be best to build a new Topology object that understands the complete topology and can route packets correctly.

dklowden, 2023-03-09:
Poking this issue as this is also of key interest for the Intel AGILE team. Is there any further ownership to enable/support this, or, is the only solution to develop a fully customized router as proposed?

mrutt92, 2024-10-23:
I see this comment is quite old. I was wondering if I wanted to do something simpler, say have many mesh networks connected through a single hr_router using a bridge, would that be possible?

mrutt92, 2024-10-23:
I have been able to have a hierarchy of single router networks.