sstsimulator/sst-elements #2063: [Merlin] How to connect hr_router to new component?

nhatdangncsu, 2023-03-06:
New Issue for sst-elements
--------------------------

1 - Detailed description of problem or enhancement
I'm trying to connect a new component to Merlin hr_router. Here 's the connection in my configuration file. Internal network is an instance of class network which contains merlin hr_router.
```
stuController = sst.Component("node"+ str(node_num) + "." +"stu_controller_" + str(next_memory_ctrl_id), "memHierarchy.STUController")
stu_cpulink = stuController.setSubComponent("cpulink", "Opal.OpalMemNIC")
stuController_ring_link = sst.Link("node"+ str(node_num) + "." +"stuController_" + str(next_memory_ctrl_id) + "_link")
stuController_ring_link.connect( (stu_cpulink, "port", ring_latency), (internal_network.rtr, "port%d"%(internal_network.getNextPort()), ring_latency))
```
Inside the STU Controller, here 's how I configure the link.
stuController.h
```
SST_ELI_DOCUMENT_SUBCOMPONENT_SLOTS(
            {"cpulink", "CPU-side link manager (e.g., to caches/cpu). Defaults to MemLink.", "SST::MemHierarchy::MemLinkBase"}
 )
MemLinkBase*            linkDown_;
```
stuController.cc
```
void STUController::configureLinks(Params &params) {
...
        linkDown_ = loadAnonymousSubComponent<MemLinkBase>("memHierarchy.MemNIC", "cpulink", 0, ComponentInfo::SHARE_PORTS | ComponentInfo::INSERT_STATS, nicParams, defaultTimeBase_);
...
}
void STUController::init(unsigned int phase) {
	**linkDown_->init(phase); //Error at this line**

	if (!phase){
		//linkDown_->sendInitData(new MemEventInitCoherence(getName(), Endpoint::Cache, type_ == "inclusive", type_ != "inclusive", cacheArray_->getLineSize()));
		linkDown_->sendInitData(new MemEventInitCoherence(getName(), Endpoint::Directory, true, true, 64, true));
    }
	while(MemEventInit *event = linkDown_->recvInitData()) {
		if (event->getCmd() == Command::NULLCMD) {
			d_->debug(_L10_, "%s received init event: %s\n",
					this->getName().c_str(), event->getVerboseString().c_str());
		}
		delete event;
	}
	return;

}
```
I 'm getting a segmentation fault at the bold line. Here 's the call stack.
```
#0  0x000000000051a49e in SST::Link::sendUntimedData (this=0x0, data=data@entry=0x7fffd57fc148) at ../../../../../sst-core/src/sst/core/link.cc:287
#1  0x00007fffd6db7d61 in SST::Merlin::LinkControl::init (this=0x12b4fc50, phase=<optimized out>) at ../../../../../../sst-elements/src/sst/elements/merlin/interfaces/linkControl.cc:206
#2  0x00007fffd662d3ce in SST::MemHierarchy::MemNIC::init (this=0x12b4e560, phase=0) at ../../../../../../sst-elements/src/sst/elements/memHierarchy/memNIC.cc:66
#3  0x00007fffd65d063d in SST::MemHierarchy::STUController::init (this=0x12b4dc40, phase=0) at ../../../../../../sst-elements/src/sst/elements/memHierarchy/stuController.cc:239
#4  0x00000000005391d7 in SST::Simulation_impl::initialize (this=0xb927f0) at ../../../../../sst-core/src/sst/core/componentInfo.h:205
#5  0x00000000004c9ea7 in start_simulation (tid=<optimized out>, info=..., barrier=...) at ../../../../../sst-core/src/sst/core/main.cc:418
#6  0x00000000004b6b61 in main (argc=<optimized out>, argv=<optimized out>) at ../../../../../sst-core/src/sst/core/main.cc:874
```
I haven't been able to identify why this error occurs. I really appreciate if someone can give me some suggestion to resolve this issue. Thanks for your help.

gvoskuilen, 2023-03-23:
@nhatdangncsu Does the STUController have an SST_ELI_DOCUMENT_PORTS() macro that declares a port named "port" in its ELI?

gvoskuilen, 2023-04-19:
@nhatdangncsu Is this still an issue?

closed 2023-07-03 by hughes-c