sstsimulator/sst-elements #2066: [Memhierarchy] Address range assignment for spad subcomponent

samkitjain, 2023-03-09:
Hello all,

I am new to SST and I would like to understand how we can assign address ranges to controller of memhierarchy:spads just like memHierarchy.MemController.

`memctrl = sst.Component("memory", "memHierarchy.MemController")
memctrl.addParams({
      "clock" : cpu_clock,
      "backend.mem_size" : "4GiB",
      "backing" : "malloc",
      "initBacking": 1,
      **"addr_range_start": 0,
      "addr_range_end": 0xffffffff,**
})
`
I am expecting SST to route request to respective spad based on the address. I would be thankful if you could point me to an example.

Thank you!

gvoskuilen, 2023-03-23:
@samkitjain The scratchpads don't have that capability and are not part of the global memH address space. We are working on adding that capability. I'll leave this issue open and give an update as soon as we have one.

josuerfi, 2023-07-05:
Do you have a timeframe when the capability might be added?