sstsimulator/sst-elements #2088: DiskSim/SSDSim

FDZX0118, 2023-03-30:
hello 

I know from section 4.3 of the paper [1] that DiskSim is integrated into the SST, but I can't find where it is and how to configure it.
Can you help me? Please

[1]"The structural simulation toolkit. SIGMETRICS Perform. Eval. Rev. 38, 4 (March 2011), 37–42. https://doi.org/10.1145/1964218.1964225"

gvoskuilen, 2023-04-04:
At one time DiskSim was integrated with SST, but it is no longer integrated or supported.

FDZX0118, 2023-04-05:
Thank you very much for your reply. Please tell me where can I find the old version of SST with DiskSim integration? Or can you share it with me? Thanks again!

hughes-c, 2023-05-09:
@FDZX0118 previous releases of sst are here: https://github.com/sstsimulator/sst-elements/releases or http://sst-simulator.org/SSTPages/SSTBuildAndInstall_older_downloads/ 

I don't see anything in the release notes on ssdsim so I don't know the last version that supported it.

closed 2023-08-14 by hughes-c