sstsimulator/sst-elements #2142: Multi-NIC

sunsirui, 2023-05-25:
Hello,
----
The EmberMPIJob constructor takes:
1 - Job ID - doesn't matter what it is, as long it is unique for each job. We make the user specify it so that messages with the job id can be mapped back to the right job
2 - Number of nodes (num_nodes) - this is actually the number of nodes, not the number of endpoints
3 - Number of cores (numCores) - this is really just the number of MPI ranks per node. Total ranks in the job will be num_nodes * numCores.
4 - Number of nics per node (nicsPerNode) - number of nics per node. The job will use num_nodes * nicsPerNode endpoints on the network. Also, the numCores must be an even multiple of nicsPerNode and each "core" is mapped to a single NIC (i.e., a single core can't use the bandwidth of multiple nics).
----
Above, the "(i.e., a single core can't use the bandwidth of multiple nics)" mentioned about the fourth parameter means that for each NIC, the MPI ranks are independent of each other after they are assigned to the NIC, that is, an MPI rank can only be assigned by its attached NICs are used to transmit data, and multiple NICs cannot send data at the same time.

That is, Multi-NIC is not equal to Multi-rail. (The Multi-rail I expect is to add a network card to the node, from the original mpi rank to send data through a single NIC to use multiple NICs to send data at the same time)

Thank you!

feldergast, 2023-05-25:
Multi-nic is equal to multi-rail with the limitation that any given rank on the node can only access one NIC.  Having 4 NICs in the simulation would correspond physically to having 4 NICs in a node, and each NIC would only service a specific set of cores/ranks.  This is how most of our large HPC clusters use multiple NICs on a node because it makes the software path more efficient, and in the case of NIC offloads, it's important that the state for a single rank is on a single NIC.  Additionally, a single thread can rarely use all the bandwidth of a single NIC and we tend to avoid MPI_THREAD_MULTIPLE for efficiency reasons. In the case of the simulation, this assumption also greatly simplified the network stack model.

From the perspective of using the NICs simultaneously, all NICs can be actively sending and receiving data at the same time, and a single NIC can simultaneously handle data from multiple ranks (as long as those ranks have been mapped to that NIC). So, it is possible in the simulation to use all the bandwidth offered by all the NICs.  Additionally, the ember/firefly stack will simulate data transferred between ranks on the same node as memory copies, and that data will not be sent out over the network.

sunsirui, 2023-05-26:
Thank you very much, I understand

closed 2023-05-26 by sunsirui