sstsimulator/sst-elements #2148: An error when run "runstream.py"

chaojibendan, 2023-06-03:
I want to run `sst runstream.py` but fail. I don‘t know why. It seems that just can't find stream? But I have set correct env `SST_ROOT`.

log is:
`memory, WARNING: Memories no longer inherit address regions from directories and no region parameters (addr_range_start, addr_range_end, interleave_size, interleave_step) were detected. All addresses will map to this memory: if this is intended, you may ignore this warning or set addr_range_start to 0 in your input deck to eliminate this warning.
[tid:74911] None : No such file or directory
FATAL: Pin3Frontend[frontend/pin3/pin3frontend.cc:373:forkPINChild] Launching trace child failed!  Child Exited with status 2
SST Fatal Backtrace Information:
    0 : sst(_ZNK3SST6Output5fatalEjPKcS2_iS2_z+0x3f7) [0x557895553657]
    1 : /home/wyy/local/sstelements-13.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend12forkPINChildEPKcPPcRSt3mapINSt7__cxx1112basic_stringIcSt11char_traitsIcESaIcEEESC_St4lessISC_ESaISt4pairIKSC_SC_EEENS0_15redirect_info_tE+0x7bd) [0x7fd6b475fdad]
    2 : /home/wyy/local/sstelements-13.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent12Pin3Frontend4initEj+0x12c) [0x7fd6b475ffbc]
    3 : /home/wyy/local/sstelements-13.0.0/lib/sst-elements-library/libariel.so(_ZN3SST14ArielComponent8ArielCPU4initEj+0x21) [0x7fd6b4719bd1]
    4 : sst(_ZN3SST15Simulation_impl10initializeEv+0x94) [0x557895577464]
    5 : sst(+0xfd8fb) [0x5578954e48fb]
    6 : sst(main+0x94d) [0x5578954c65dd]
    7 : /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0xf3) [0x7fd6b76f3083]
    8 : sst(_start+0x2e) [0x5578954e38de]
--------------------------------------------------------------------------
MPI_ABORT was invoked on rank 0 in communicator MPI_COMM_WORLD
with errorcode 1.

NOTE: invoking MPI_ABORT causes Open MPI to kill all MPI processes.
You may or may not see output from other processes, depending on
exactly when Open MPI kills them.
--------------------------------------------------------------------------`

plavin, 2023-06-05:
`runstream.py` expects the executable name to be in the environment variable `OMP_EXE`. 

Try running `OMP_EXE=./stream sst runstream.py`.

chaojibendan, 2023-06-06:
Thank you!  Now I can run `runstream.py`.

closed 2023-06-06 by chaojibendan